raspberrypi/pico-sdk #1084: Calling multicore_lockout_start_blocking() causes: assertion "us_since_boot <= INT64_MAX" failed

tjko, 2022-10-30:
On "Debug" builds calling _multicore_lockout_start_blocking()_ triggers assertion:

```
assertion "us_since_boot <= INT64_MAX" failed: file "/home/tjko/pico/pico-sdk/src/common/pico_base/include/pico/types.h", line 63, function: update_us_since_boot
```

(cannot reproduce this if CMAKE_BUILD_TYPE is set to "Release")

Trace from gdb:
```
Thread 1 received signal SIGTRAP, Trace/breakpoint trap.
_exit (status=status@entry=1) at /home/tjko/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:183
183	        __breakpoint();
(gdb) where
#0  _exit (status=status@entry=1) at /home/tjko/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:183
#1  0x10008f9a in __assert_func (file=file@entry=0x1002e048 "/home/tjko/pico/pico-sdk/src/common/pico_base/include/pico/types.h", 
    line=line@entry=63, func=func@entry=0x10033558 <__func__.4243> "update_us_since_boot", 
    failedexpr=failedexpr@entry=0x1002e02c "us_since_boot <= INT64_MAX") at /home/tjko/pico/pico-sdk/src/rp2_common/pico_runtime/runtime.c:228
#2  0x1000e4fc in update_us_since_boot (us_since_boot=<optimized out>, t=<synthetic pointer>)
    at /home/tjko/pico/pico-sdk/src/common/pico_base/include/pico/types.h:63
#3  delayed_by_us (us=9223372036840512308, t=...) at /home/tjko/pico/pico-sdk/src/common/pico_time/include/pico/time.h:101
#4  make_timeout_time_us (us=9223372036840512308) at /home/tjko/pico/pico-sdk/src/common/pico_time/include/pico/time.h:130
#5  multicore_fifo_push_timeout_us (data=data@entry=1940423454, timeout_us=<optimized out>)
    at /home/tjko/pico/pico-sdk/src/rp2_common/pico_multicore/multicore.c:34
#6  0x1000e63e in multicore_lockout_handshake (magic=magic@entry=1940423454, until=...)
    at /home/tjko/pico/pico-sdk/src/rp2_common/pico_multicore/multicore.c:212
#7  0x1000e6de in multicore_lockout_start_block_until (until=...) at /home/tjko/pico/pico-sdk/src/rp2_common/pico_multicore/multicore.c:235
#8  0x1000e8a0 in multicore_lockout_start_blocking () at /home/tjko/pico/pico-sdk/src/rp2_common/pico_multicore/multicore.c:246
#9  0x10003d3c in save_config () at /home/tjko/git/fanpico/src/config.c:688
#10 0x10001136 in cmd_save_config (cmd=<optimized out>, args=<optimized out>, query=<optimized out>, prev_cmd=<optimized out>)
    at /home/tjko/git/fanpico/src/command.c:246
#11 0x1000293e in run_cmd (cmd=cmd@entry=0x20041ba0 "conf", cmd_level=0x20000994 <config_commands>, cmd_level@entry=0x200008a4 <commands>, 
    prev_subcmd=prev_subcmd@entry=0x20041b58) at /home/tjko/git/fanpico/src/command.c:1370
#12 0x100029da in process_command (state=state@entry=0x200032c4 <system_state>, config=<optimized out>, 
    command=command@entry=0x20041ba0 "conf") at /home/tjko/git/fanpico/src/command.c:1406
#13 0x10000a5a in main () at /home/tjko/git/fanpico/src/fanpico.c:349
(gdb) 

```

tjko, 2022-10-30:
This is probably same/duplicate as issue #1059

kilograham, 2022-10-30:
closing as duplicate short of any further reproduction steps

closed 2022-10-30 by kilograham