raspberrypi/pico-sdk #1103: Please, Clean SDK

Wiz-IO, 2022-11-18:
Each end user SDK version must be complete ( no submodule links ) and cleaned of unnecessary files
eg: effective code is 6M vs ~60M .git vs ~76M for full folder

at the very least, the download speed increases
At least i think so...

JamesH65, 2022-11-18:
git clone --depth?

Or have I misunderstood the question?

Wiz-IO, 2022-11-18:
yep, there are options and solutions (by default)
my thought is that there are a lot of junk files for the end user
and I thought a simple repo (end version) optimization wouldn't hurt
I look Kilogram has prepared mbedtls, which will further increase the final transfer

Wiz-IO, 2022-11-18:
[image]

aallan, 2022-11-18:
Sorry, not entirely sure what the question is?

Wiz-IO, 2022-11-18:
no question, just an optimization tip

peterharperuk, 2022-11-18:
Which files do you think are unnecessary?

Wiz-IO, 2022-11-18:
for the end user, only SRC & LIB are effective, maybe DOC too
for what is the student's 60M dot git folder ( with the entire history of the repo ) 
there is always a dev repo

and source codes for ESP32, TI etc...

Wiz-IO, 2022-11-18:
here's a question for you
for me, the PICO-SDK installation is 20 seconds( ~80 megabytes, effective 7 )
how much is it with you?
and how much is per board RPI?

JamesH65, 2022-11-18:
I just did a git clone https://github.com/raspberrypi/pico-sdk.git and it took about 3s. (This is my home net connection, not in the office)

What do you mean by "and how much is per board RPI?"

Wiz-IO, 2022-11-18:
3 sec, nice
Raspberry Pi 4

JamesH65, 2022-11-18:
The SDK tree seems to be about 13MB accord to "du".

Do you mean how long it takes to download on a Pi4? About the same, 2.5s

Wiz-IO, 2022-11-18:
ok

closed 2022-11-18 by Wiz-IO

peterharperuk, 2022-11-18:
Presumably you have to add the time to execute something like "git submodule update --init --depth 1"?

JamesH65, 2022-11-18:
So the submodule stuff took 9.6s on the Pi4.

Bumped size on disk up to 41MB.

kilograham, 2022-11-18:
note #1044 will fix the biggest offender

lurch, 2022-12-29:
Note that if you want to download the source code for the SDK without getting the `.git` folder too, you can download one of the zip files from https://github.com/raspberrypi/pico-sdk/releases
`pico-sdk-1.4.0.zip` is only 1.2MB.

Wiz-IO, 2022-12-29:
I needed it for automatic installation and updates - PlatformIO / Pico SDK the **full** version 
with git install/update...
other solutions become complicated