raspberrypi/pico-sdk #1106: lwip-mqtt: sys_timeout: timeout != NULL, pool MEMP_SYS_TIMEOUT is empty #1

pingufreak, 2022-11-20:
Hi all,

I'm using pico_cyw43_arch_lwip_threadsafe_background with the lwip mqtt library. Anyways, I'm getting:

`*** PANIC ***

sys_timeout: timeout != NULL, pool MEMP_SYS_TIMEOUT is empty`

The implementation is based on pico-sdk/lib/lwip/contrib/examples/mqtt/mqtt_example.c :

`int main() {
 stdio_init_all();
 if(cyw43_arch_init_with_country(CYW43_COUNTRY_GERMANY)) {
  printf("Failed to initialise cyw43_arch\n");
 }
 cyw43_arch_enable_sta_mode();
 if(cyw43_arch_wifi_connect_timeout_ms(WIFI_SSID, WIFI_PASSWORD, CYW43_AUTH_WPA2_AES_PSK, 10000)) {
  printf("Failed to connect to WiFi\n");
 }
 mqtt_example_init();
 while(true) {
  printf("loop\n");
  sleep_ms(1000);
 }
 return 0;
}`

I've also implemented it from scratch with the documentation  `pico-sdk/lib/lwip/doc/mqtt_client.txt` with the same result.

I've also tested the implementation of https://github.com/cniles/picow-iot with the same result.

Do you have an idea? Maybe there are issues with the sys_timers of the library itself? 

Is the mqtt-library only supported with FreeRTOS or also working with pico_cyw43_arch_lwip_threadsafe_background?

Thank you!

Kr

pingufreak

pingufreak, 2022-11-20:
I've increased LWIP_NUM_SYS_TIMEOUT_INTERNAL and it works now.

kilograham, 2022-11-24:
assuming this can be closed?

albertasgalin, 2023-02-15:
Thanks! [pingufreak](https://github.com/pingufreak) it realy helped!

closed 2023-02-15 by kilograham

ruswerner, 2023-11-19:
@pingufreak What was the value you used? and what specific define did you set?

I ended up getting it working by doing this in my header:

```
#define MEMP_NUM_SYS_TIMEOUT (LWIP_NUM_SYS_TIMEOUT_INTERNAL + 100)
```

... however I was just guessing at the value and setting it too high produces a compilation failure due to binary size too large to fit in RAM.

ruswerner, 2023-11-19:
I think I found my answer by searching the `pico_sdk` folder...  There is a txt file in `<SDK>/lib/lwip/doc/mqtt_client.txt` with this note at the top:

```
You need to increase MEMP_NUM_SYS_TIMEOUT by one if you use MQTT!
```

I feel like this should be more explicitly addressed in the code examples!

lurch, 2023-12-20:
Note that anything in `<SDK>/lib/lwip/` is coming from https://github.com/lwip-tcpip/lwip rather than directly from pico-sdk itself.

OKuvshinov, 2025-05-29:
> I think I found my answer by searching the `pico_sdk` folder... There is a txt file in `<SDK>/lib/lwip/doc/mqtt_client.txt` with this note at the top:
>
> ```
> You need to increase MEMP_NUM_SYS_TIMEOUT [...]

I use SNTP and Wireguard. Increasing this value from 3 to 6 helps me too