raspberrypi/pico-sdk #1107: Disable LWIP_DHCP not possible, because DHCP is hardcoded in the SDK

pingufreak, 2022-11-20:
Hi all,

I've searched a solution to get native static ip addresses supported without any use of DHCP. DHCP is using time, energy and computing. Therefore it makes sense, to have an option to define static addresses. LWIP supports setting static addresses via net_if. Is there a plan to have an option to enable / disable DHCP and use of manual static addresses?

Thx

Kr

pingu

peterharperuk, 2022-11-21:
No plan, but it sounds like an issue we should fix.

brgalo, 2022-12-05:
Also ran into the same issue.
I am not very experienced with the entire issue/pullrequest workflow, but i would offer to help, if i get pointed in the right direction.

peterharperuk, 2022-12-05:
I believe this is a driver issue, see https://github.com/georgerobotics/cyw43-driver/issues/48