raspberrypi/pico-sdk #1108: Debug raspberrypi pico using openocd's daplink error

wangwang105, 2022-11-21:
When I use the openocd command to debug raspberrypi pico, I get an error:
openocd configuration:

rp2040.svd
cmsis-dap.cfg
rp2040.cfg
arm-none-eabi-gdb.exe
adapter speed 1000

```
Open On-Chip Debugger 0.12.0-rc2 (2022-10-26-14:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

CDRTOSConfigure
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: FW Version = 1.10
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: CMSIS-DAP command 0x1d not implemented
Error: CMSIS-DAP command SWD_Sequence failed.
Info : Read incorrect DLPIDR 0xf0000001 (possibly CTRL/STAT value)
Error: Failed to connect multidrop rp2040.dap1
```
Hoping to get some answers,thank you！

kilograham, 2022-11-24:
Please take questions here https://forums.raspberrypi.com/viewforum.php?f=143

closed 2022-11-24 by kilograham

leitcode, 2023-01-25:
@wangwang105 I have exactly the same issue, did you already find a solution?

nickfox-taterli, 2024-08-04:
> @wangwang105 I have exactly the same issue, did you already find a solution?

When we use Linux x64, do not use the openocd branch of the Raspberry Pi can solve this issue, of course, JLink debugging experience is better, except a little expensive.

peterharperuk, 2024-10-16:
No and you're unlikely to get any update on a closed issue. The original issue doesn't appear to have the full details of how to reproduce the problem, what command line was used etc