raspberrypi/pico-sdk #1118: Dont exclude baudrate BOOTSEL reset trick from build when LIB_TINYUSB_HOST or LIB_TINYUSB_DEVICE is defined

popy2k14, 2022-11-25:
As the title says and discussed here: https://github.com/raspberrypi/pico-sdk/issues/838#issuecomment-1325602575

Please do not exclude function tud_cdc_line_coding_cb() from file /rp2_common/pico_stdio_usb)/reset_interface.c
The function is very helpful to bring the pi into BOOTSEL mode when TinyUSB is used.
A better approach would be to include it always and make an define to exclude it, when not needed.

As a workaround, i have just copied, adapted and inserted the function into my project, like this

```
// Support for default BOOTSEL reset by changing baud rate
void tud_cdc_line_coding_cb(__unused uint8_t itf, cdc_line_coding_t const* p_line_coding) {
    if (p_line_coding->bit_rate == 110) {    
        //board_led_write(1);
        reset_usb_boot(gpio_mask, 0);
    }
}
```

kilograham, 2024-07-21:
We will now respect PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE=1 if you set it when using TinyUSB yourself (it won't default to 1 like it does when you aren't using TinyUSB)

kilograham, 2024-08-09:
This should work fine with SDK 2.0.0

closed 2024-08-09 by kilograham

lurch, 2025-12-15:
> but `picotool info` doesn't see any compatible devices that it can force into bootsel mode

Correct, I believe that `picotool` only knows about the `PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE` and knows nothing about the `PICO_STDIO_USB_ENABLE_RESET_VIA_BAUD_RATE`.

> I'm sure I'm missing something very basic and I'm going to feel dumb for not realizing it but I can't see it. What am I doing wrong?

This is the sort of question that you ought to be asking on the [forums](https://forums.raspberrypi.com/viewforum.php?f=143), rather than on GitHub.