raspberrypi/pico-sdk #1128: Can we have an option for CYW43 Worker task stack size?

ronter-7786, 2022-12-02:
Currently it uses configMINIMAL_STACK_SIZE. The only other use for this is the IDLE task(s) that need very little stack. I have found that CYW43 Worker task needs more than 300 entries. Seems a waste of RAM for 2 IDLE tasks to take so much stack as well.

peterharperuk, 2022-12-02:
Yes - I think I've seen issues with stack size of the worker thread as well. I assume a different build macro would do the job?

ronter-7786, 2022-12-02:
Yes, thank you. A build macro as there is for CYW43_TASK_PRIORITY would be great!
 If not specified, then the default could still be configMINIMAL_STACK_SIZE.

peterharperuk, 2023-01-23:
This will now be resolved by https://github.com/raspberrypi/pico-sdk/pull/1177

peterharperuk, 2023-01-25:
Should be fixed now see CYW43_TASK_STACK_SIZE

closed 2023-01-25 by peterharperuk