raspberrypi/pico-sdk #1129: Allow changing the USB vendor ID, product ID, manufacturer, and product name using build settings

theacodes, 2022-12-02:
Presently, it's not possible to change the USB identifiers without forking the SDK and modifying `src/rp2_common/pico_stdio_usb/stdio_usb_descriptors.c`. This can be a challenge when shipping a product or even just developing multiple RP2040-based devices at the same time, since it's very difficult to figure out which device is which when multiple are connected.

This ideally should be possible to change through `CMakeLists.txt` like this:

```Cmake
target_compile_definitions(project PUBLIC USBD_VID=0xCAFE USBD_PID=0xCAFE USBD_MANUFACTURER="Widgets, Inc" USBD_PRODUCT="Widget")
```

lurch, 2022-12-02:
On a related tangent, see also https://github.com/raspberrypi/usb-pid

kilograham, 2022-12-03:
Merged into develop

closed 2022-12-03 by kilograham