raspberrypi/pico-sdk #1131: printf %a formatting just generates an "a"

steverpalmer, 2022-12-03:
The C99 option of "%a" formatting for floating point numbers is just generating an "a", so, i guess, is not being recognised as a formatting option.

I've just tested this on:
- Debian 11.5 (fresh build in a VM);
- used the `pico-setup.sh` configuration script;
- tweaked the serial hello-world example to `printf("Hello, world! %a\n", 0.1);`
- rebuild and load onto a fresh PicoW (using uf2 image);
- outputs "Hello, world! a".

This may be something to do with Marco Paland's `printf`, which doesn't appear to support the %a option.  However, I've had no success if I tweak the build system to use arm-none-eabi-newlib (v3.3.0).

Or perhaps I am missing something?

kilograham, 2022-12-03:
depends what you tried!

```
pico_set_printf_implementation(your_target compiler)
```

steverpalmer, 2022-12-06:
Thanks @kilograham for the suggestion, but despite 30 years as a professional software engineer with half being in embedded systems, CMake still confuses me.  I couldn't see how to use `pico_set_printf_implementation` to force the system to use the newlib `printf`, so I resorted to butchery ;-).

- In the sdk I removed directories `pico_printf`, `pico_stdio`, `pico_stdio_semihosting`, `pico_stdio_uart`, `pico_stdio_usb` from `src/rp2_common`;
- I patched the CMake errors (commenting out lines in `rp2_common/CMakeLists.txt` and `rp2_common/pcio_runtime/CMakeLists.txt` that referred to the missing directories); 
- My test project C source is just `int main() { printf("%a\n", 0.1); }`;
- My test project CMakeList.txt does not use `pico_enable_stdio_` and merely has `target_link_libraries(test pico_runtime)`;
- I added the necessary `libgloss` function, most trivially returning an error, but also with a simple `_write` to send the bytes through `uart0`;
- Stepping through the program execution, I can see that it is using newlib's `printf`.

The result is that it still ignores the `%a` formatting!  Further investigation suggests that the default build of newlib does not include C99 IO formatting options.  It needs to be configured with `--enable-newlib-io-c99-formats` to add `%a` formatting.

I guess the real problem is that there is insufficient interest in the `%a` feature to make folk demand it (at the expense of more code).  My interest comes from my attempt to port Lua (which by default does use `%a`) to the Pico.  However, it also looks like Lua source has a way to work around this, so maybe I'll tackle it there...

For what its worth, if you think that this "issue" serves no purpose, I am content to see it closed.  Cheers.

steverpalmer, 2022-12-06:
Aside: running the Lua testsuite has just thrown another oddity - `0.0f / 0.0f` on the Pico (with a fresh, unbutcherred SDK) is returning `INFINITY` rather than `NAN`.  I can raise this as a distinct issue if you want, or it might not be worth further consideration.

rblank, 2023-04-06:
Just to confirm after testing, and for posterity:

- Setting `pico_set_printf_implementation(your_target compiler)` fixes the issue (i.e. `%a` becomes available), but only if newlib is built with `--enable-newlib-io-c99-formats`.
- The alternative for Lua is to undefine `lua_number2strx` in `luaconf.h`.

steverpalmer, 2023-04-17:
Hi @rblank,
You are right that Lua's configuration uses the function `lua_number2strx` to convert a float to a hexadecimal, and by making it undefined, Lua provides its own implementation in `lstrlib.c`.  Controlling `lua_number2strx` is the most general approach to solving such number formatting issues. (For posterity ;-) I solved the problem using the specific build flag `LUA_NO_C99_FORMATTING=1` which seems to be intended for just this case and (as you suggest) leaves `lua_number2strx` undefined in `luaconf.h`.
Cheers.

closed 2023-04-17 by steverpalmer