raspberrypi/pico-sdk #1132: pico_get_unique_board_id: Unique id is not always unique

ThinkTransit, 2022-12-04:
I have 2 Pico W boards that return the same unique id with pico_get_unique_board_id()

For Pico W devices this can be resolved by combining with the mac address to form a unique combination, not sure what options exist for non wireless version.

liamfraser, 2023-05-31:
@ThinkTransit do you have any more detail here? Was this a flash binary or no flash? No flash binaries are not expected to have a unique ID. On RP2040 the unique ID comes from the flash chip. I assume they are not sharing the same flash chip?

ThinkTransit, 2023-06-02:
@liamfraser I flashed the boards with the [Pimoroni](https://github.com/pimoroni/pimoroni-pico) micropython firmware. Boards aren't using the same flash chip.

Looking through the micropython code it calls pico_get_unique_board_id().

I checked all my boards and it is only 2 boards that have the same unique_id so maybe it is a freak occurrence?

liamfraser, 2024-08-01:
I think this might just be a freak occurrence. Potentially we could mix in the WiFi mac address on Pico W @peterharperuk, although that would require initializing the wireless driver at start of day which is probably undesirable.

ThinkTransit, 2024-08-02:
@liamfraser thanks yes it must be, I haven't seen it happen with any other boards. Maybe I just won the UUID lottery.

Will reopen it I come across it again.

closed 2024-08-02 by ThinkTransit

Tonykids, 2024-08-13:
I bought 50 [luatos-rp2040](https://cdn.openluat-luatcommunity.openluat.com/attachment/20230814165827547_luatos-rp2040.pdf) boards 6 monthon ago for my project  and find they return the same unique id.
finally I find that it uses [PY25Q32](http://www.junmintech.com/UploadFiles/Product/20231128173412_79023.pdf) instead of normal W25Q32 .
PY25Q32 has **128bit** unique ID,this may be the problem.
[image]
[image]

lurch, 2024-08-13:
@Tonykids See #1641

peterharperuk, 2024-08-13:
@Tonykids Any idea where I can get one of those boards for testing a fix for this?

lurch, 2024-08-15:
@Tonykids You might like to test #1819 ?