raspberrypi/pico-sdk #1140: Error when linking agasit mbedtls "fatal error: lwip/opt.h: No such file or directory"

LOCNNIL, 2022-12-10:
I'm trying to use Mbed-TLS library on Raspberry Pi Pico board. But after following the instructions in @peterharperuk [commit](https://github.com/raspberrypi/pico-sdk/commit/719d5e41a2f9c9347e43563b6bbcb2c46f780d65#diff-602b4198c72c0254ecd8a40b7da19b9c3bef56055792026a9fc1ed59fc5ae9fa)  (450a4d5) that fixes #894 I'm noting beeing able to compile my aplication when linking with `pico_lwip_mbedtls` and `pico_mbedtls`. 

- **The error**
```bash
[build] /home/lincoln/pico/pico-sdk/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c:52:10: fatal error: lwip/opt.h: No such file or directory
[build]    52 | #include "lwip/opt.h"
[build]       |          ^~~~~~~~~~~~
[build] compilation terminated.
[build] gmake[2]: *** [CMakeFiles/crypto_lock.dir/build.make:1059: CMakeFiles/crypto_lock.dir/home/lincoln/pico/pico-sdk/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls.c.obj] Error 1
[build] gmake[2]: *** Waiting for unfinished jobs....
[build] [ 24%] Building C object CMakeFiles/crypto_lock.dir/home/lincoln/pico/pico-sdk/lib/lwip/src/apps/snmp/snmpv3_mbedtls.c.obj
[build] /home/lincoln/pico/pico-sdk/lib/lwip/src/apps/altcp_tls/altcp_tls_mbedtls_mem.c:51:10: fatal error: lwip/opt.h: No such file or directory
[build]    51 | #include "lwip/opt.h"
```

On my linking libraries I've selected:
```cmake
    target_link_libraries(${PROJECT_NAME} 
        pico_stdlib 
        pico_stdio 
        hardware_uart 
        hardware_irq
        pico_lwip_mbedtls
        pico_mbedtls
        )
```
If I remove `pico_mbedtls` and  `pico_lwip_mbedtls` it compiles fine.

**NOTE**: I'm not using any functionality of mbedtls yet. I'm just trying to check if it works. 

**NOTE 2**: My system infos:

```bash
Linux pop-os 6.0.6-76060006-generic #202210290932~1669062050~22.04~d94609a SMP PREEMPT_DYNAMIC Mon N x86_64 x86_64 x86_64 GNU/Linux

cmake version 3.22.1
```

LOCNNIL, 2022-12-10:
EDIT: I'm using the last version of `pico-sdk` [version 1.4.0](https://github.com/raspberrypi/pico-sdk/releases/tag/1.4.0), on [commit](https://github.com/raspberrypi/pico-sdk/commit/2e6142b15b8a75c1227dd3edbe839193b2bf9041)[2e6142b](https://github.com/raspberrypi/pico-sdk/commit/2e6142b15b8a75c1227dd3edbe839193b2bf9041).

LOCNNIL, 2022-12-27:
I solved this issue by cross-compiling MbedTLS myself and linking it against my generated .uf2 file, as I explained in [this issue in the MbedTLS repository.](https://github.com/Mbed-TLS/mbedtls/issues/6763#issuecomment-1345438369)

underhood, 2023-03-04:
seem to be having exact same issue. Is there a way to make it work with pico-sdk? (without compiling own MbedTLS?)

peterharperuk, 2023-03-06:
Let's see if we can make it more obvious. And you should be able to use mbedtls without lwip?

LOCNNIL, 2023-03-06:
@peterharperuk In most part I think that yes. Because, acording with my researches **lwIP** is a "open-source TCP/IP stack designed for embedded systems" right? But in my case I wish to use `mbedtls` for radio ( `RF` ) aplications and not TCP/IP or other socket aplication, my goal was basically ensure a secure **encrypted** RF transmission.

peterharperuk, 2023-03-06:
Correct. I've only every used it with an IP stack but I'd expect it to be possible.

kilograham, 2023-03-06:
um a bit confused: yes we should be able to link against mbedtls without lwip, but why are you including both 

```
 pico_lwip_mbedtls
 pico_mbedtls
```

?

if you don't want lwip, just use pico_mbedtls (no i haven't tried it, but your error above comes from trying to compile pico_lwip_mbedtls)

LOCNNIL, 2023-03-06:
@kilograham What you said seems very logic, I agree with you, but when I just tried to add `pico_mbedtls` at `target_link_libraries` seems that the linker couldn't find the library:

```cmake
[build] Scanning dependencies of target crypto_lock
[build] Consolidate compiler generated dependencies of target crypto_lock
[build] [ 20%] Linking CXX executable crypto_lock.elf
[build] /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: cannot find -lpico_mbedtls: No such file or directory
[build] collect2: error: ld returned 1 exit status
[build] gmake[2]: * [CMakeFiles/crypto_lock.dir/build.make:1449: crypto_lock.elf] Error 1
[build] gmake[1]: * [CMakeFiles/Makefile2:1491: CMakeFiles/crypto_lock.dir/all] Error 2
[build] gmake: * [Makefile:91: all] Error 2
[proc] The command: /usr/bin/cmake --build /home/lincoln/TFG/lock/build --config Debug --target all -j 6 -- exited with code: 2
[build] Build finished with exit code 2
``` 

I think this linking error is because I'm doing something wrong.

About my setup: I check in git log and I'm currently in  the latest commit at master [f396d05](https://github.com/raspberrypi/pico-sdk/commit/f396d05f8252d4670d4ea05c8b7ac938ef0cd381)

closed 2023-05-16 by peterharperuk