raspberrypi/pico-sdk #1147: add_repeating_timer_us only being called once

ygator, 2022-12-16:
The code below is the callback for a timer that is setup with:
add_repeating_timer_us(-16670, repeating_timer_callback, NULL, &timer);

When BAD is defined the callback only gets called once and does not repeat.  When BAD is not defined it functions correctly.

Does anyone else see similar results?  What could the issue be?

Thanks,

Offending Code

```
#define BAD

volatile uint8_t tick = 0;
volatile uint8_t rollover = 0;

bool repeating_timer_callback(struct repeating_timer *t) {

	rollover = 1;
	tick++;
#ifdef BAD
	// using this code this callback only gets called once
	if((tick % 10) > 5) {
#else
	// using this code this callback get called repeatedly
	if(tick > 5) {
#endif
		rollover++;
		tick=0;
	}
}

```

Complete files

test.cpp

```
#include <pico/stdlib.h>
#include <stdio.h>
#include <tusb.h>

const uint LED_PIN = PICO_DEFAULT_LED_PIN;
struct repeating_timer timer;

bool repeating_timer_callback(struct repeating_timer*);
void test(void);

int main(int argc, char *argv[]) {

	stdio_init_all();

	// Configure LED
	gpio_init(LED_PIN);
	gpio_set_dir(LED_PIN, GPIO_OUT);

#if 1
	// blink LED until USB connected for stdout
	while(!tud_cdc_connected()) {
		sleep_ms(100);
		gpio_put(LED_PIN, 1 - gpio_get(LED_PIN));
	}
	printf("USB is now connected\n");
        gpio_put(LED_PIN, 1);
#endif

	test();

	printf("end of main\n");
}

//#define BAD
volatile uint8_t tick = 0;
volatile uint8_t rollover = 0;

bool repeating_timer_callback(struct repeating_timer *t) {

	gpio_put(LED_PIN, 1 - gpio_get(LED_PIN));

	rollover = 1;
	tick++;
#ifdef BAD
	// using this code this callback only gets called once
	if((tick % 10) > 5) {
#else
	// using this code this callback gets called repeatedly
	if(tick > 5) {
#endif
		rollover++;
		tick=0;
	}
}

void test(void) {

	add_repeating_timer_us(-16670, repeating_timer_callback, NULL, &timer);

	while(1) {
		sleep_ms(1000);
		printf("1s\n");
	}

	printf("end of test\n");
}

```
CMakeLists.txt

```
cmake_minimum_required(VERSION 3.12)

# Pull in SDK (must be before project)
include(pico_sdk_import.cmake)

project(test C CXX ASM)
set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

# Initialize the SDK
pico_sdk_init()

add_executable(test
	test.cpp
)

# Pull in basic dependencies
target_link_libraries(test pico_stdlib hardware_spi hardware_pwm)

# create map/bin/hex file etc.
pico_add_extra_outputs(test)

# Link to pico_stdlib (gpio, time, etc. functions)
target_link_libraries(${PROJECT_NAME}
    pico_stdlib
)

# Enable USB output, disable UART output
pico_enable_stdio_usb(${PROJECT_NAME} 1)
pico_enable_stdio_uart(${PROJECT_NAME} 0)

```

ygator, 2022-12-16:
Thanks, I missed the return value.  Copied from code that worked fine that also did not return a value.

Will use forums next time.

BTW, the build showed no warnings.  Will need to look to see how to enable them.

Thanks again.

kilograham, 2022-12-16:
> BTW, the build showed no warnings. Will need to look to see how to enable them

Ah, i didn't realize they were off by default

You can either use `add_compile_options(-Wall)` in your CMakeLists.txt or use the target specific `target_compile_options`