raspberrypi/pico-sdk #1148: Bug / Issue with frequency_count_khz

tcroswell, 2022-12-17:
I believe there is a small bug in frequency_count_khz.

When utilizing the pico examples "...clocks/hello_48mhz", the code will hang in the function "measure_freqs".  The call that is hanging is the call to frequency_count_khz(uint...).

In tracing thru that call, there is a statement:
"
   while(fc->status & CLOCKS_FC0_STATUS_RUNNING_BITS) {
        tight_loop_contents();
    }
"
which, in my code will hang and never return.

The statement: "(fc->status & CLOCKS_FC0_STATUS_RUNNING_BITS)" evaluates to TRUE and thus the code is locked in a tight loop.

In looking at it more closely and in testing it out in my own version, I believe the code should be:
while(!(fc->status & CLOCKS_FC0_STATUS_RUNNING_BITS))

I apologize if I've done something incorrectly, but the example code does not work at all on my system.

Thanks

Tom

bruelltuete, 2023-02-14:
can you elaborate which call to measure_freqs() hangs, there are two. also which call to frequency_count_khz(), because there are a few.
The while loop as it is in the function is fine https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_clocks/clocks.c#L222: "for as long as the status bit says that the frequency counter is busy spin in this loop".

there could be a minor bug in the hello_48MHz as it does not flush the uart fifo when changing clk_peri, so you could end up with garbled characters in the console.

tcroswell, 2023-02-14:
Hi;

Thanks very much for contacting me.  Unfortunately, the error was all mine and you can close this out.  I was using similar (but not identical) hardware which was not behaving as expected to the code I identified below.  I didn't realize that the differences (though minor) would cause this behavior.  Once I realized my mistake, I was able to get it to work.  Just close out this bug report and blame it on an idiot developer.  I apologize that I forgot to reach back to the bug submission and close this out for you.  I hope you did not waste too much time with this.

Tom

closed 2023-05-26 by kilograham