raspberrypi/pico-sdk #1153: Ubuntu: undefined reference to `_exit' - collect2: error: ld returned 1 exit status

SciLor, 2022-12-23:
I have set up everything with the script named here: https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

But I get this error when trying to build the examples on a fresh Ubuntu Container:

Linux CT100 5.15.39-3-pve #2 SMP PVE 5.15.39-3 (Wed, 27 Jul 2022 13:45:39 +0200) x86_64 x86_64 x86_64 GNU/Linux
Distributor ID: Ubuntu
Description:    Ubuntu 22.04 LTS
Release:        22.04
Codename:       jammy

```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /bin/arm-none-eabi-gcc 
Build flags: ;-mcpu=cortex-m0plus;-mthumb
Id flags:  

The output was:
1
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(lib_a-exit.o): in function `exit':
/build/newlib-pB30de/newlib-3.3.0/build/arm-none-eabi/thumb/v6-m/nofp/newlib/libc/stdlib/../../../../../../../../newlib/libc/stdlib/exit.c:64: undefined reference to `_exit'
collect2: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /bin/arm-none-eabi-g++ 
Build flags: ;-mcpu=cortex-m0plus;-mthumb
Id flags:  

The output was:
1
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/lib/thumb/v6-m/nofp/libc.a(lib_a-exit.o): in function `exit':
/build/newlib-pB30de/newlib-3.3.0/build/arm-none-eabi/thumb/v6-m/nofp/newlib/libc/stdlib/../../../../../../../../newlib/libc/stdlib/exit.c:64: undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```

kilograham, 2022-12-23:
please ask for help with build setup etc. here https://forums.raspberrypi.com/viewforum.php?f=143

closed 2022-12-23 by kilograham

SciLor, 2022-12-23:
This is no build setup problem. This error shouldn't occur. I can build everything afterwards.

lurch, 2022-12-24:
I've just followed the instructions in Chapter 2 and Chapter 3 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf myself in a fresh Ubuntu 22.04 Docker image, and when I run the cmake step I see:
```
# cmake ..
Using PICO_SDK_PATH from environment ('../../pico-sdk')
PICO_SDK_PATH is /root/pico/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from /root/pico/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 
TinyUSB available at /root/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
cyw43-driver available at /root/pico/pico-sdk/lib/cyw43-driver
lwIP available at /root/pico/pico-sdk/lib/lwip
-- Configuring done
-- Generating done
-- Build files have been written to: /root/pico/pico-examples/build
```

So I'm afraid I can only conclude that you must have done something wrong somewhere? :man_shrugging: 

So try again, and if you're still having problems, ask for help on the forums.

SciLor, 2022-12-24:
It works yes, and I can make after that. But there is an error withing the CMakeError.log
This hints that there is a problem.