raspberrypi/pico-sdk #1156: Is cyw43 driver leaving DMA channels in some undefined state?

bruelltuete, 2022-12-25:
The cyw43 driver uses DMA to feed the PIO machines, e.g. https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/cyw43_driver/cyw43_bus_pio_spi.c#L275

But does it need to do more to clean up on deinit? See https://github.com/raspberrypi/pico-sdk/blob/develop/src/rp2_common/cyw43_driver/cyw43_bus_pio_spi.c#L169

I've noticed that when I deinit the wifi module, my ADC code will re-use one of the DMA channels previously used by the cyw driver. And that re-used DMA channel will pretty much immediately trigger a completion IRQ even when I haven't started the ADC nor the channel itself. So something was left pending.

Should there be an `dma_irqn_acknowledge_channel(DMA_IRQ_0, bus_data->dma_out);` just before dma_channel_unclaim? (same for `DMA_IRQ_1` and `dma_in`).

peterharperuk, 2023-05-12:
I can reproduce this. Not sure I understand where it's coming from as we don't use interrupts for dma. I presume it doesn't matter if we use DMA_IRQ_0 or DMA_IRQ_1 as both channels can use the same interrupt. A call to dma_irqn_acknowledge_channel for both channels we use does seem to fix it.

kilograham, 2023-06-02:
fix merged into develop

PR #1372 (2023-06-02): Add dma_channel_cleanup method and use it to cleanup after pico_cyw43_driver is closed
cyw43 seems to be causing a dma channel interrupt to go off if the same channels are reused after cyw43 is deinitialised. Achnowledging them seems to fix it.

Fixes #1156
closed 2023-06-02 by kilograham