raspberrypi/pico-sdk #1157: SVD file is missing USB host-only registers and fields

yodaldevoid, 2022-12-26:
The SVD file is missing a register and a field at the moment that are only used for USB host operation. Both overlap with device-only registers/fields.

The register is EPx control. It can be found at USB DPSRAM offset 0x100, or the bottom of table 393 in the RP2040 datasheet (p.g. 386). This overlaps with the device-only EP0 buffer 0.

The field doesn't really have a name concrete name, but it is called interrupt interval in the TinyUSB code. It can be found in bits 16:25 of the endpoint control registers, or table 394 of the RP2040 datasheet (p.g. 386). This overlaps with the interrupt on stall and interrupt on NAK fields.

lurch, 2022-12-26:
> Both overlap with device-only registers/fields.

Yeah IIRC we only document the device-mode USB registers in the SVD file, because I don't think there's any way in SVD to specify "overlapping" registers? ping @liamfraser

yodaldevoid, 2022-12-26:
There is currently no register defined at an offset of `0x100` for the USBCTRL_DPRAM peripheral or any register at an equivalent address in another peripheral so there is no problem of overlap there.

In the CMSIS-SVD spec I don't see explicit support for overlapping fields. If overlapping fields are not supported, separate register definitions for device and host operation could be defined and paired using the `alternateRegister` element.

lurch, 2022-12-27:
Yes, looks like that might work https://arm-software.github.io/CMSIS_5/SVD/html/elem_registers.html

ping also @peterharperuk as he was looking at our SVD generator most recently.