raspberrypi/pico-sdk #1160: picoasm and gnu assembler

HondaRulez, 2022-12-27:
HI
An option would be fine to generate an assembler compatible .h file
or a #ifdef to exlude the C specific lines. (to keeping the public #defines only)

lurch, 2022-12-27:
Some of the header files **do** already do that, e.g.
https://github.com/raspberrypi/pico-sdk/blob/develop/src/common/pico_base/include/pico/config.h#L11
https://github.com/raspberrypi/pico-sdk/blob/develop/src/boards/include/boards/pico.h#L8

Which `.h` files do you want to use in assembler, that you're not able to do at the moment?

HondaRulez, 2022-12-27:
The picoasm generated .h file. Currently I defined some pins in the *.pio asm because can't "include" mine "hwpins.h", I don't want to define twice the pins (what can lead some trouble if I forget something :D ) so the pin what are defined in the *.pio asm I using in  the C code and also want use in the asm file. currently the assembler don't like the "wrap" definitios. the other stuff can be excluded by #define PICO_NO_HARDWARE, mine current solution is wrote a C file what defines some static structures with the defs/labels in the generated pio.h file. then those structures referenced in the asm code.

visenri, 2025-03-09:
Given the original author intend:

> Currently I defined some pins in the *.pio asm because can't "include" mine "hwpins.h"

Using "pico_generate_pio_header_pre" as I've implemented here:
https://github.com/visenri/pico_cmake_extras
allows the use of all c preprocessor features.

So it can be used to include other files or use preprocessor macros defined elsewhere.

Not exactly the request of this topic, but it is an alternative that IMO offers a better way to do exactly what the author intended.

@kilograham:
This should be easy to integrate into the next SDK release.

kilograham, 2025-07-22:
@will-v-pi I had missed the linked new CMake function; can't make it in 2.2.0 but lets add it to the next release