raspberrypi/pico-sdk #1162: Cannot create a new section modifying the default linker script.

shripathi-kamath, 2023-01-01:
I do not know if this is the right forum, but I tried the raspberry pi forums to no avail.

My requirement is to have a portion of the RAM reserved for holding code created at runtime, and for the section to be writable.  Plus, I want to load some of its content by copy a portion of the FLASH at boot-up.  The purpose is to have dynamic generation of code.

I have a minimal sample that is modeled after the USB stdio sample for the SDK examples.  It simply prints a string.  I have attempted to create a modified linker script (based on the default script) as follows:

main.cpp:
```
#include <stdio.h>

extern "C" void forth();

int main()
{
    forth();
    return 0;
}

```
forth() is defined in an assembly file called test.s

test.s:
```
.syntax unified
.thumb

.section .data
.align 2
data: .asciz "ABC12345"

.section .user_section, "awx" @ putting function in .user_section
@ .section .text
.align 2
.global forth 
.thumb_func
forth:
    PUSH    {LR}
    BL      stdio_init_all
    LDR     R0, =data
    BL      puts
    POP     {PC}
```

The linker file has the following mods:
```
MEMORY
{
    FLASH(rx) : ORIGIN = 0x10000000, LENGTH = 2048k
    RAM(rwx) : ORIGIN =  0x20000000, LENGTH = 128k
    FORTH(rwx) : ORIGIN =  0x20020000, LENGTH = 128k
    SCRATCH_X(rwx) : ORIGIN = 0x20040000, LENGTH = 4k
    SCRATCH_Y(rwx) : ORIGIN = 0x20041000, LENGTH = 4k
}
```

and the section definition of user_section as follows:

```
.user_section : {
        *(.user_section.*)
        . = ALIGN(4);
    } > FORTH AT> FLASH
```

The .dis file looks like this:
```
Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .boot2        00000100  10000000  10000000  00001000  2**0
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .text         000071e8  10000100  10000100  00001100  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .rodata       000016b0  100072e8  100072e8  000082e8  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  3 .binary_info  00000020  10008998  10008998  00009998  2**2
                  CONTENTS, ALLOC, LOAD, DATA
  4 .ram_vector_table 000000c0  20000000  20000000  0000b038  2**2
                  CONTENTS
  5 .data         00000460  200000c0  100089b8  0000a0c0  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  6 .uninitialized_data 00000000  20000520  20000520  0000b0f8  2**0
                  CONTENTS
  7 .user_section 00000038  20020000  10008e18  0000b000  2**3
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  8 .scratch_x    00000000  20040000  20040000  0000b0f8  2**0
                  CONTENTS
  9 .scratch_y    00000000  20041000  20041000  0000b0f8  2**0
                  CONTENTS
 10 .bss          00000d84  20000520  20000520  0000b520  2**3
                  ALLOC
 11 .heap         00000800  200012a4  200012a4  0000b0f8  2**2
                  CONTENTS, READONLY
```

The .dis file appears to be correct, but the program does not run.  If I change the section to just .text (see comment in test.s), it works fine.  If I additionally use the copy_to_ram linker script distributed with the SDK, the code is loaded in RAM on boot. What I want is a combination of the two, in that user_section should be copied from FLASH to ram and should allow for additional code to be placed at runtime.

I think I am doing all the right things but am stumped as to what I am missing.  Any suggestions on making this work are appreciated.

shripathi-kamath, 2023-04-10:
Does anyone has any suggestions?

kilograham, 2023-04-11:
Sorry; somehow never saw this. Questions are better asked on the forums:

https://forums.raspberrypi.com/viewforum.php?f=143

closed 2023-04-11 by kilograham

shripathi-kamath, 2023-04-11:
Thank you.  I had posted it on the forum, and then here:
https://forums.raspberrypi.com/viewtopic.php?p=2065927#p2065927

Shripathi

kilograham, 2023-04-12:
ok; yeah hopefully someone responds there, otherwise I will try to.