raspberrypi/pico-sdk #1164: Bluetooth Support for Pico W

kilograham, 2023-01-03:


tannewt, 2023-01-04:
Will you be using NimBLE for the host-side?

kilograham, 2023-01-05:
Our integration uses [BTstack](https://github.com/bluekitchen/btstack)

arturo182, 2023-01-05:
I'm no lawyer but isn't this worrying in their license?

https://github.com/bluekitchen/btstack/blob/master/LICENSE#L20

```
Redistribution and use in source and binary forms, with or without
modification, are permitted provided that the following conditions
are met:

[...]

4. Any redistribution, use, or modification is done solely for
   personal benefit and not for any commercial purpose or for
   monetary gain.
```

kilograham, 2023-01-05:
It will be licensed by us for more permissive use on RP2040

arturo182, 2023-01-05:
Thanks for the clarification 👍

WayneKeenan, 2023-01-06:
Looking forward to this. Thanks.

marian-m12l, 2023-01-06:
This is great news :+1: 
Just to manage expectations, could you clarify the scope for the 1.5.0 target? BLE? Dual mode? Maybe support for some profiles? Device and Host?
Thanks!

JamesH65, 2023-01-07:
I'm not sure we will be publishing that in advance, but I will try and find out.

cezarfloroiu, 2023-02-06:
Any updates on when Bluetooth support will be released?

JamesH65, 2023-02-06:
I believe this should be fairly soon.

gnbl, 2023-02-06:
> Any updates on when Bluetooth support will be released?

4 days ago, according to the [1.5.0 milestone](https://github.com/raspberrypi/pico-sdk/milestone/14)  ;-)

znmeb, 2023-02-06:
> > Any updates on when Bluetooth support will be released?
> 
> 4 days ago, according to the [1.5.0 milestone](https://github.com/raspberrypi/pico-sdk/milestone/14) ;-)

Time flies when your teeth are blue ;-)

kilograham, 2023-02-06:
> > > Any updates on when Bluetooth support will be released?
> > 
> > 
> > 4 days ago, according to the [1.5.0 milestone](https://github.com/raspberrypi/pico-sdk/milestone/14) ;-)
> 
> Time flies when your teeth are blue ;-)

literally from the cold, and the power being out

SpeedyCraftah, 2023-02-06:
Enough of the chatter lads, time to emit some radio waves 🌊 📻

peterharperuk, 2023-02-07:
The gears are turning https://github.com/georgerobotics/cyw43-driver/pull/60

BenRoe, 2023-02-09:
Any changes on the PCB needed, to support Bluetooth with the Pico W?
Or is it only a software thing?

kilograham, 2023-02-11:
Beta support released into SDK 1.5.0 and `pico-examples` 

see https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.0

closed 2023-02-11 by kilograham

carlosdelfino, 2025-04-23:
Hello, is there any tutorial in C/C++ that demonstrates how to use Bluetooth with the RP Pico W? Has BTstack been released for commercial use? Thanks.

lurch, 2025-04-23:
@carlosdelfino See https://github.com/raspberrypi/pico-examples/?tab=readme-ov-file#pico-bluetooth and https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_btstack/LICENSE.RP