raspberrypi/pico-sdk #1165: Long WiFi connect times when in polling mode

dylanhall, 2023-01-05:
I've been trying to resolve an issue where the time my Pico W takes to connect to the WiFi varies a lot.
I'm testing with the following code:

```
#include <stdio.h>
#include "pico/stdlib.h"
#include "pico/cyw43_arch.h"

int main() {
    stdio_init_all(); // using UART for stdio
    printf("Initialising WiFi hardware\n");
    if (cyw43_arch_init()) {
	printf("Failed to initialise wifi hardware\n");
	return 0;
    }
    cyw43_arch_enable_sta_mode();
    cyw43_wifi_pm(&cyw43_state, CYW43_DEFAULT_PM & ~0xf);
    printf("Connecting to WiFi\n");
    uint64_t connect_time = time_us_64();
    while (cyw43_arch_wifi_connect_timeout_ms(WIFI_SSID, WIFI_PASSWORD, CYW43_AUTH_WPA2_AES_PSK, 30000)) {
	printf("Failed to connect, trying again %llu ms\n", (time_us_64() - connect_time)/1000);
    }
    printf("Connected %llu ms\n", (time_us_64() - connect_time)/1000);
}
```
When using _pico_cyw43_arch_lwip_threadsafe_background_ the connect times are typically 3-5 seconds, with the occasional issue like this:
```
Initialising WiFi hardware                                                      
Connecting to WiFi                                                              
Failed to connect, trying again 3240 ms                                         
Connected 8207 ms
```
That is pretty much a worst case example, so about 8 seconds.

However, when using _pico_cyw43_arch_lwip_poll_ the connect times vary far more with a worst case of over a minute.
I think the issue is in _cyw43_arch.c_ in the function _cyw43_arch_wifi_connect_until_.
It looks like _best_effort_wfe_or_timeout_ doesn't wake up frequently enough or at the right times so _cyw43_arch_poll_ isn't called often enough to complete the necessary handshakes in a timely manner. To try and prove this I modified cyw43_arch_wifi_connect_until to count the number of times it loops. e.g.
```
int cyw43_arch_wifi_connect_until(const char *ssid, const char *pw, uint32_t auth, absolute_time_t until) {
    int err = cyw43_arch_wifi_connect_async(ssid, pw, auth);
    if (err) return err;

    int status = CYW43_LINK_UP + 1;
    int poll_count = 0;
    while(status >= 0 && status != CYW43_LINK_UP) {
        int new_status = cyw43_tcpip_link_status(&cyw43_state, CYW43_ITF_STA);
        if (new_status != status) {
            status = new_status;
            CYW43_ARCH_DEBUG("connect status: %s\n", status_name(status));
        }
        // in case polling is required
	cyw43_arch_poll();
	poll_count++;
        best_effort_wfe_or_timeout(until);
        if (time_reached(until)) {
            printf("Poll count %d\n", poll_count);
            return PICO_ERROR_TIMEOUT;
        }
    }
    printf("Poll count %d\n", poll_count);
    return status == CYW43_LINK_UP ? 0 : status;
}
```
A couple example runs:
```
Initialising WiFi hardware                                                      
Connecting to WiFi                                                              
Poll count 24                                                                   
Connected 25490 ms

Initialising WiFi hardware                                                      
Connecting to WiFi                                                              
Poll count 4                                                                    
Failed to connect, trying again 3286 ms                                         
Poll count 2                                                                    
Failed to connect, trying again 6185 ms                                         
Poll count 26                                                                   
Connected 19402 ms
```
Those poll counts seem low unless there is some way to guarantee they occurred at just the right time.
I initially tried adding a _printf_ to the loop but that generated hundreds of lines of output and seemed to resolve the issue.
The next experiment was to comment out _best_effort_wfe_or_timeout_ which also resolved the issue. I assume this is bad for power management however.

I note this is the same area of code that is mentioned in issue 917 (https://github.com/raspberrypi/pico-sdk/issues/917). 

Thanks!

kilograham, 2023-01-23:
this may well be fixed by #1177

kilograham, 2023-01-24:
#1177 is merged into develop; please re-open if that does not fix the issue

closed 2023-01-24 by kilograham