raspberrypi/pico-sdk #1168: gcc's optimise for size broken

bruelltuete, 2023-01-08:
I've tried to build a pico project with optimise-for-size.
One particular function has an integer division in it.
gcc generated a call to divmod_s32s32.
So far so good.

divmod_s32s32 clobbers r3, as is allowed. https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_divider/divider.S#L107

but gcc assumes r3 is preserved.

Does divmod_s32s32 need some more annotation re calling convention?

Only happens with optimise-for-size. With optimise-for-speed it uses a different register and r3's value doesn't matter.

gcc version is 9.2.1, pretty old... i think i followed some how-to for installing this when i first started playing with the Pico...

kilograham, 2023-01-08:
Weird, can you attach a repro, and/or say whether it is fixed in later GCC

bruelltuete, 2023-01-08:
will try to compile a recent gcc from source. i have no idea how to work linux package management to get a recent binary into wsl...

dhalbert, 2023-01-08:
You can download recent arm-none-eabi-gcc versions from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads. Just unpack them and put the executable directory on your path.

bruelltuete, 2023-01-09:
ah thanks for the link! that makes my life much easier...

12.2.Rel1 from arm's website does not show this problem. I've tried a few different combinations of global (cmdline flags) and local (function attribute) optimisation flags and the disassembler output looks fine. And the impacted function produces the correct result.

12.2 is a lot more bloaty though: bin size is 400k vs 340k for 9.2.

closed 2023-01-09 by bruelltuete