raspberrypi/pico-sdk #1169: Allow flash programming of partial pages (<256 bytes)

hattesen, 2023-01-09:
https://github.com/raspberrypi/pico-sdk/blob/2e6142b15b8a75c1227dd3edbe839193b2bf9041/src/rp2_common/hardware_flash/flash.c#L86

taraskornuta, 2023-02-06:
Also want to support this meaning. Such an approach will also reduce the lifetime of the flash itself.

hattesen, 2023-02-15:
> Also want to support this meaning. Such an approach will also reduce the lifetime of the flash itself.

Allowing partial page writes will not, in any way, reduce the lifetime of the flash. It is page erasures that wear out flash, not page writes (which should obviously not be overlapping). For an effective EEPROM simulation in Flash, allowing partial page writes will reduce the number of flash erase operations required, and thus reduce the wear of the flash, increasing the lifetime of it.

lurch, 2023-02-15:
I guess adding such an API / function would introduce the possibility of trying to overwrite some already-written data (i.e. not freshly-erased space), and thus introduces the possibility that what you _thought_ you'd written into flash, won't be the same as what you actually read back from flash? :thinking: 

I know it's probably not the easiest solution, but instead of trying to write partial-pages <256 bytes, how hard would it be to modify your application to buffer write-data in RAM until you have a full page-worth to be written to flash?

peterharperuk, 2023-02-15:
Sounds reasonable to me to add an API to do this. It could be made optional returning an error if the flash didn't support partial page writes (is that even a thing?).

hattesen, 2023-02-15:
@peterharperuk i have looked at a LOT of serial Nor flash datasheets, including those used by Adafruit, Pimoroni, seed, sparkfun and waveshare (many of which uses slow generic SPI access at half speed, compared to the "original" Winbond W25QxxJV), and they all seem to support partial page writes. They may differ in "wrap-at-end-of-page" behavior, but it would be reasonable for a partial page programming function to restrict writes to a single page without exceeding the end, typically causing wrap to beginning of (256 byte) page anyway.

The existing `flash_range_program()` function could easily be extended to support `flash_offset` outside page boundaries if `count` is less than 256, raising an error if `count` bytes will not fit in the page pointed to by `flash_offset`.

hanan619, 2025-05-06:
Any update on this issue?