raspberrypi/pico-sdk #1170: Option PICO_RP2040_B0_SUPPORTED in boards files is ignored

pdg137, 2023-01-13:
In pico.h, if I replace these lines

https://github.com/raspberrypi/pico-sdk/blob/1.4.0/src/boards/include/boards/pico.h#L80-L82

with this:

```
//#ifndef PICO_RP2040_B0_SUPPORTED
#define PICO_RP2040_B0_SUPPORTED 0
//#endif
```

and try to build Micropython, I get the following error:

```
In file included from /home/paul/gitprojects/public/micropython/ports/rp2/build-PICO/generated/pico_base/pico/config_autogen.h:7,
                 from /home/paul/gitprojects/public/micropython/lib/pico-sdk/src/common/pico_base/include/pico/config.h:19,
                 from /home/paul/gitprojects/public/micropython/lib/pico-sdk/src/common/pico_base/include/pico.h:29,
                 from /home/paul/gitprojects/public/micropython/lib/pico-sdk/src/rp2_common/pico_printf/include/pico/printf.h:45,
                 from /home/paul/gitprojects/public/micropython/lib/pico-sdk/src/rp2_common/pico_printf/printf.c:38:
/home/paul/gitprojects/public/micropython/lib/pico-sdk/src/boards/include/boards/pico.h:81: error: "PICO_RP2040_B0_SUPPORTED" redefined [-Werror]
   81 | #define PICO_RP2040_B0_SUPPORTED 0
      | 
In file included from /home/paul/gitprojects/public/micropython/lib/pico-sdk/src/rp2_common/pico_printf/printf.c:37:
/home/paul/gitprojects/public/micropython/lib/pico-sdk/src/rp2_common/pico_platform/include/pico/platform.h:43: note: this is the location of the previous definition
   43 | #define PICO_RP2040_B0_SUPPORTED 1
      | 
cc1: all warnings being treated as errors
make[3]: *** [CMakeFiles/firmware.dir/build.make:4763: CMakeFiles/firmware.dir/home/paul/gitprojects/public/micropython/lib/pico-sdk/src/rp2_common/pico_printf/printf.c.obj] Error 1
make[2]: *** [CMakeFiles/Makefile2:1383: CMakeFiles/firmware.dir/all] Error 2
make[1]: *** [Makefile:91: all] Error 2
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:33: all] Error 1
```

It appears that [platform.h sets the value](https://github.com/raspberrypi/pico-sdk/blob/1.4.0/src/rp2_common/pico_platform/include/pico/platform.h#L43) of this option before the boards file has a chance to define it, at least the way it's included from printf.c.  It looks like most of the boards are attempting to set that option to zero, but it won't do anything.

kilograham, 2023-01-13:
Yes, it seems that that file and a couple of others, for some likely historical reasons include "pico/platform.h" directly, which is wrong.

kilograham, 2023-01-24:
merged into develop

PR #1190 (2023-01-24): Remove (legacy) direct inclusion of 'pico/platform.h' and 'pico/config.h'
Now `pico.h` is safe to include from everywhere (including assembly files), and ensures that defines and configs are created in the right order

fixes #1170
fixes #950
closed 2023-01-24 by kilograham