raspberrypi/pico-sdk #1176: Kconfig interface for Pico SDK (WiP)

DatanoiseTV, 2023-01-18:
I am currently hacking on Kconfig support for my custom Pico SDK template, but would love to see something like this "out of the box" in the PicoSDK. It provides some options that are also passed by the pico-project-generator usually:

peterharperuk, 2023-01-18:
A lot of the config options *should* be marked with 25,4: // PICO_CONFIG: <name>, <description>, group=<LIB> I wonder if it could use those?

lurch, 2023-01-18:
> A lot of the config options should be marked with // PICO_CONFIG: , , group= I wonder if it could use those?

Yup, those automatically get extracted by https://github.com/raspberrypi/pico-sdk/blob/master/tools/extract_configs.py into the `pico_configs.tsv` used by https://github.com/raspberrypi/pico-project-generator/
ping @JamesH65 for his thoughts.

DatanoiseTV, 2023-01-18:
Also as side-note things like GPIO and custom peripheral initialization and CPU overclocking could be maybe done via:
```c
void __attribute__ ((constructor)) premain(){
    // init hardware and overclock cpu
}
```
to be done before main() gets executed, so it is more pleasant to the user.

DatanoiseTV, 2023-01-19:
With the current proof of concept, i am able to load defconfigs for different boards which can also contain pin definitions for custom peripheral etc. It would be great if the pico-sdk would adopt the possibility to include "components" like in the esp-idf which can contain their own Kconfig. 

For building I can use now:
```shell
cmake -GNinja -DBOARD=datanoise_picoadk ..
ninja
```

For config I can use now:
```shell
cmake -GNinja -DBOARD=datanoise_picoadk ..
ninja menuconfig
```