raspberrypi/pico-sdk #1182: iCache is borderline useless as it operated by default. Please provid einfo on any chicken bits to improve it

dgrinbe2, 2023-01-20:
Why implement a fancy QSPI unit and then handicap it to the tune of this much lost perf?
The whole premise of cache is that if you used bit n and n plus one you will likely need n plus two. So making a tiny cache line is just silly.
RP2040 uses a very strange cache geometry which effectively 4BYTE lines (lines are 8 but filled one half at a time). No bursts for fill ever. I wish you'd consider a change long-term.
You can increase perf by over 55% by just forcing the current cache to do full line fills and not half like you do (details later)...

The cache has eight byte lines, but you allow each line to be half filled. I have never seen it do a full line fill for data access (it does do it for code sometimes).
But there is a reason that everyone else in the world uses 16, 32, or 64 byte lines. 
I understand why you made this choice (you think this will improve latency, and someone will care), but you are just wrong.
NOBODY places latency sensitive code in flash in RP2040 due to the unexpected (and punishingly high) latencies of a cache miss anyways.
So you really should say f*** this idea and go with longer lines.

The overhead of setting up a read REALLY kills you!
Even in "skip command QPI mode", half of every transaction is address!

You could, in fact, more than double the throughout. At 66MHz DDR QPI, the flash is 533Mbit/s!! (assuming long enough lines to use that)
As you use it, you get under half that. One cycle between transactions, 3 for address, 6 for dummy, 4 for data now. So you are using this bus to a 28% efficiency at best (4/(4+6+1+3))

With a 32 byte line that goes up to 76.2% (32/(32+6+1+3))
So 3x the BW
Three f***ing x!!

In any case, I assume there are no chicken bits for changing cache line size (this sort of thing tends to be hard wired in silicon), but surely there must be one to force full linefills in the current cache? Always filling 8 bytes should still provide for somewhat better bandwidth: (8/(8+6+1+3)) = .44 (44% of max theoretical bandwidth, 1.55x current BW).
Why do I suspect that a chicken bit must exist? Because nobody else makes caches with split lines like this (for a good reason), this must have been a "ooh, i am so very very clever" idea from someone, and surely some other level-headed grey-bearded engineer sighed, said "ok" but left a bit in the RTL to disable this nonsense.

Maybe you'll share the bit number and register address for this bit?
Please?

daveythacher, 2023-01-20:
It cost's a dollar. Honestly the flash is for booting RAM. **Cache lines matter more for hit-miss cache!** This cache is not really able to use a prefetch as this would drive cost too high and latency would explode. Therefore you are stuck with cache line size.

You are correct the XIP cache is very small for the performance of the XIP flash. In order to get high bus utilization you would need to increase the cache line size. In order for the cache's locality to be meaningful you would need a certain number of cache lines. This will drive area and mostly waste it considering the latency. Therefore you get more main RAM for the programmer to figure it out.

Your points in general are correct; however I think these points should be considered.

dgrinbe2, 2023-01-20:
As the lines are currently 8 bytes, but some fills only fill half of a line (sometimes first half, sometime second), is there a setting to always fill the entire line? It is certainly possible because sometimes it is observed.

dgrinbe2, 2023-01-20:
> Honestly the flash is for booting RAM

If this were the intended use case, why is there a cache at all? The expected hit rate of a cache with no prefetching in the states situation (a one time load of data from A to B) is precisely 0% (every single access will miss). Surely there is more to this?

daveythacher, 2023-01-21:
We can produce requests faster than we can fill them. So we save the old answers. We have a random answer pool converted into an ordered request. This is computationally expensive in software.

Sometimes we need to preempt the current request and fill it later. This happens if we have too many requests, which is possible on single core? If we branch the engine may need to abort it's current effort, which is wasteful? We have low hit rate in this case so we issue many bad requests. Yet we keep moving at a faster clock speed because we abort. 

Our fill time is faster because we use more bandwidth on productive fills. (Even if we waste time.) We would love to be in linear execution of fetch. Yet we are temporal, because we do not have the resources for prefetch. We would likely want to be super-scalar and/or speculative for that to even make sense? We use coherency in RAM to avoid problems?

Our fill time is so low there is no point to anything complex on flash. Only RAM could fill fast enough to avoid the pipeline from being stalled. Deep pipelines would stall hard without speculation. Ironically this is why high clock cores are basically microprocessors rather than microcontrollers. Microprocessors with low latency IO are starting to appear to fill the void. RP2040 is actually a simplified model of this.

daveythacher, 2023-01-21:
Compute the instruction throughput. Number of instructions used vs number of instructions fetched. Multiply by number of fetches possible per second. Vary the number of instructions used. You will see that this predicts a different reality. 

No matter what you choose you will find that one program will respond better than another!

dgrinbe2, 2023-01-21:
An instruction during, say, memcpy, can load 28 bytes, and those 28 bytes are seen on the bus as 7x 4byte reads, not 3x8 byte reads + 4x byte read. So the question is: is there some bit somewhere, to cause data fill to fill the entire 8-byte line at once? It could cost latency but provide more throughput for tasks that are BW bound

Wren6991, 2023-01-23:
Hi [dgrinbe2](https://github.com/dgrinbe2),

> The cache has eight byte lines, but you allow each line to be half filled. I have never seen it do a full line fill for data access (it does do it for code sometimes).

You're correct, the Synopsys SSI only supports a single preconfigured transfer size, and for XIP this must be no larger than the APB data bus width of 32 bits. The unusual cache line layout (8 bytes with an upper/lower valid bit) is a workaround for this. Issuing an 8-byte fill from the cache would just trigger two 4-byte accesses from the SSI, which would _not improve throughput_ due to the operation of the SSI.

Due to the small fill size there is little exploitation of spatial locality, but in any code with reasonable temporal locality (e.g. bulk code in protocol stacks) we see good uplift from the inclusion of the cache.

> The overhead of setting up a read REALLY kills you!
Even in "skip command QPI mode", half of every transaction is address!

Yes, we have done similar back of the envelope calculations and reached the same conclusions as you. The use of the SSI was an engineering compromise, and it's one of our highest priorities to improve in future devices.

Note for long linear reads it is possible to achieve 100% throughput via direct SSI DMA, and there is example code for this [here.](https://github.com/raspberrypi/pico-examples/blob/master/flash/ssi_dma/flash_ssi_dma.c)

> In any case, I assume there are no chicken bits for changing cache line size (this sort of thing tends to be hard wired in silicon), but surely there must be one to force full linefills in the current cache?

Your assumption is correct, there are no such chicken bits. Because the limitation is in the fixed transfer size of the SSI, manipulating the downstream accesses from the cache wouldn't increase the performance. This is a fairly fundamental limitation of the SSI (it's a QSPI peripheral with XIP functionality grafted onto it) and reworking it would be around as much effort as writing our own QSPI memory interface from scratch.

I'll take this issue as feedback that full-line fills and chaining of QSPI transfers across sequentially addressed fills are strongly desired features. We also don't like to leave wasted cycles on the floor.

closed 2023-01-23 by aallan