raspberrypi/pico-sdk #1191: Is there an SVD definition file for RP2040 Pico-w?

Joebeazelman, 2023-01-25:
I am updating an existing RP2040 HAL written in Ada to support the Pico-w. Unfortunately, I can't find any definitions in the RP2040 SVD for it. Is it defined in another SVD, or is it forthcoming?

JamesH65, 2023-01-25:
The SVD file applies to the RP2040 chip itself, so since the PicoW uses the same chip it should use the same SVD file. The wireless functionality is supplied by a chip connected by SPI to the RP2040 and is not part of the RP2040 itself.

lurch, 2023-01-25:
Just to add to what James said, board-specific differences are covered by the files in https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards

Joebeazelman, 2023-01-26:
Oops, I didn't mean the chip itself, but the Pico board. It's hard to get the terminology correct. I was under the false impression that the RP Pico _board_ had its own SVD file. In reality, only the RP2040 has one. I guess SVDs are not intended for describing boards.

lurch, 2023-01-27:
> I guess SVDs are not intended for describing boards.

Correct. [SVD files](https://www.keil.com/pack/doc/CMSIS/SVD/html/index.html) are for describing "the system contained in Arm Cortex-M processor-based microcontrollers, in particular, the memory mapped registers of peripherals."

kilograham, 2023-02-09:
there is just one for RP2040

closed 2023-02-09 by kilograham