raspberrypi/pico-sdk #1196: Pre/Post build actions

taraskornuta, 2023-01-25:
Hello. Try to trigger some post-build actions, but without luck. To be specific, want to show information about the occupied resources. It can be done by executing the command in terminal "arm-none-eabi-size --format=berkeley --totals app.elf"

I'm new with CMake and it looks that I something missing. I try to add post-build command:
add_custom_command(TARGET ${ELF_FILE} POST_BUILD
         COMMAND arm-none-eabi-size --format=berkeley --totals "${CMAKE_CURRENT_BINARY_DIR}/src/${PROJECT_NAME}.elf")
I put this into the end of root CMakeLists.txt
But it doesn't show anything after the build is finished.
Can someone explain to me what I'm doing wrong?

implodnik, 2023-02-14:
The way it works for me:

```
add_custom_command(TARGET ${TARGET_NAME}
	POST_BUILD
	COMMAND ${CMAKE_COMMAND} -E echo ""
	COMMAND ${PICO_GCC_TRIPLE}-size --format=gnu $<TARGET_FILE_NAME:${TARGET_NAME}>
	COMMAND ${CMAKE_COMMAND} -E echo ""
	)
```
${TARGET_NAME} is defined by me and contains the target name. Echos are just for adding empty lines before and after the table.

taraskornuta, 2023-02-14:
@implodnik Thanks, it works!

closed 2023-02-14 by taraskornuta