raspberrypi/pico-sdk #1218: Automatic Ejection of Raspberry Pi Pico W after Upload in mac

Vishnuxx, 2023-02-04:
I'm having trouble debugging my **Raspberry Pi Pico W** via Arduino Serial monitor. Every time I upload my program, the Pico gets automatically ejected from my mac and the serial monitor gets disconnected. I've tried with multiple ports but the issue still persists. The program is running fine on the Pico, but the disconnection makes it difficult to debug.

### Environment: 
**Operating System:** `macOs Ventura 13.0 `
**IDE:** `Arduino IDE 2.0 `
**Board:** ` Raspberry Pi Pico W `
**Board Manager:**` Raspberry Pi Pico / RP2040 by Earle F. Philhower, lll version 2.7.1 `

### Details: 
The program runs fine on the Pico but gets ejected from the system after uploading. This issue occurs even when testing with a Blink example code from the Arduino IDE.

### Steps to Reproduce: 

1.  Connect the Raspberry Pi Pico W in BOOTSEL Mode.
2. Make sure to check RPI-RP2 drive folder in the Finder.
3. Choose Raspberry Pi Pico W board from Boards Manager and Select the port (here i have tried `/dev/cu.wlan-debug Serial Port` and `UF2 Board UF2 Devices`)
4. Blink example from `File>Examples>Basics>Blink
5. Write some code to Serial print
6. Compile and Upload

### Output

   ```Sketch uses 335572 bytes (16%) of program storage space. Maximum is 2093056 bytes.
   Global variables use 68896 bytes (26%) of dynamic memory, leaving 193248 bytes for local variables. Maximum is 262144 
   bytes.
   Resetting /dev/cu.wlan-debug
   Converting to uf2, output size: 702976, start address: 0x2000
   Flashing /Volumes/RPI-RP2 (RPI-RP2)
   Wrote 702976 bytes to /Volumes/RPI-RP2/NEW.UF2
   ```

At the same time Pico gets ejected automatically. And No Logs can be seen in Serial Monitor.

lurch, 2023-02-04:
I think this is expected behaviour? It sounds like you're connecting to the USB-stdio port on the RP2040, and this is implemented by software running on the RP2040. So when you upload new code to your Pico W, the RP2040 automatically resets (so that the new code can start running) and this means that the USB-stdio code on the RP2040 (perhaps only temporarily) stops running, and this causes your computer to think that the USB-stdio port has been "unplugged".

AFAIK the only way around this is to use a hardware-UART port instead via a separate USB-UART adaptor, see section 9.1.4.1 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

Vishnuxx, 2023-02-04:
Hii , Using hardware-UART may work but I'm currently using both pins for I2C connection and I'm still looking to get the output through the built-In USB of pico instead of using the hardware-UART.

Someone also has this similar issue.
https://forum.arduino.cc/t/after-i-upload-a-code-pico-disconnects/1086017

here 
  ```
   If you want output from the Pico, then your code needs to call Serial.begin() to set up the USB port on the Pico as a 
   serial interface to the computer
  ```

This is expected to work but unfortunately it didn't

lurch, 2023-02-04:
If you're using the Arduino environment, then you ought to be asking for help on the Arduino GitHub / Arduino forums, rather than on the Raspberry Pi GitHub, as Arduino is not directly supported by Raspberry Pi.

Vishnuxx, 2023-02-04:
Ok Thanks I'm closing this issue here

closed 2023-02-04 by Vishnuxx