raspberrypi/pico-sdk #1238: watchdog_enable() confuses milliseconds and microseconds

awabom, 2023-02-10:
In the file:

https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_watchdog/include/hardware/watchdog.h

delay_ms is described as:

"delay_ms Number of milliseconds before watchdog will reboot without watchdog_update being called. Maximum of 0x7fffff, which is approximately 8.3 seconds"

But 0x7fffff is 8388607 which is 8.3 seconds only if the unit is microseconds. The text also mentions "then the delay_ms parameter will not be in microseconds".

So I guess it is microseconds? Then I also think the parameter should be named delay_us, not delay_ms.

lurch, 2023-02-10:
Hmmm, https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/hardware_watchdog/watchdog.c#L55 does
```
        // Note, we have x2 here as the watchdog HW currently decrements twice per tick
        load_value = delay_ms * 1000 * 2;

        if (load_value > 0xffffffu)
            load_value = 0xffffffu;
```
so it looks like the documentation is correct in describing the parameter as milliseconds (with the hardware internally using microseconds), but it's the maximum-value which is wrong? As the parameter **is** in milliseconds, I guess the actual max-value is 8388. (which gives a timeout only 1215 microseconds less (i.e. 0.1ms) than the watchdog's maximum timeout)

awabom, 2023-02-11:
(which gives a timeout only 1215 microseconds less (i.e. 0.1ms) than the watchdog's maximum timeout)

Sorry, but 1215 microseconds is 1.2 milliseconds... :)

I haven't tested the actual delay, but indeed the parameter looks to be milliseconds, and the maximum value was wrong.

lurch, 2023-02-11:
> Sorry, but 1215 microseconds is 1.2 milliseconds... :)

LOL of course - must have been end-of-the-week brain :laughing:   Have edited my previous comment, thanks.

lurch, 2023-06-01:
As pointed out in #1409 this affects the `watchdog_reboot` function too.

kilograham, 2023-12-15:
merged into develop

PR #1567 (2023-12-15): Add watchdog parameter-validation, and fix up misleading comments
Fixes #1238 and #1550 (and #1241 #1409 #1505 which are duplicates of #1238)
closed 2023-12-15 by kilograham