raspberrypi/pico-sdk #1240: How to send HCI command to bluetooth controller ?

phb98, 2023-02-10:
Hello. AFAIK the cyw firmware already support bluetooth, only the software on host side does not support, so at this moment how can i send and receive raw HCI data to bluetooth controller ?
The bluedroid bluetooth stack on ESP32 is already mature, so i can port the stack from esp-idf

peterharperuk, 2023-02-10:
I think the answer is, wait not very long and you can look at the code yourself. Duplicate of https://github.com/raspberrypi/pico-sdk/issues/1164

peterharperuk, 2023-02-13:
You should be able to update to the latest version of the SDK v1.5 now.

To send a HCI command call [cyw43_bluetooth_hci_write](https://github.com/georgerobotics/cyw43-driver/blob/9bfca61173a94432839cd39210f1d1afdf602c42/src/cyw43_ctrl.c#L780). It requires a 4-byte header. The first 3 bytes of this header are filled in for you. The final byte in the header is the hci packet type. The buffer must be 4-byte aligned. See [hci_transport_cyw43_send_packet ](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_cyw43_driver/btstack_hci_transport_cyw43.c#L87) for an example of how it's used with BTstack.

When there's some BT work to be done [cyw43_bluetooth_hci_process ](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_cyw43_driver/btstack_hci_transport_cyw43.c#LL145C6-L145C33) gets called by the driver. [cyw43_bluetooth_hci_read ](https://github.com/georgerobotics/cyw43-driver/blob/9bfca61173a94432839cd39210f1d1afdf602c42/src/cyw43_ctrl.c#L765) reads hci data. Again it has a 4 byte header of which probably only the final byte giving the hci packet type will be of interest, see [hci_transport_cyw43_process ](https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_cyw43_driver/btstack_hci_transport_cyw43.c#LL128C13-L128C40) for an example.

kilograham, 2023-02-13:
not really an issue, so closing

closed 2023-02-13 by kilograham