raspberrypi/pico-sdk #1243: add_repeating_timer_us

globalkrishna, 2023-02-11:
add_repeating_timer_us(-1250, xx, NULL, xx) works well with pico but fails with pico-w

kilograham, 2023-02-11:
you're going to have to be a bit more specific

globalkrishna, 2023-02-12:
My code in pico was toggling on board LED in the service routine using
gpio_put(PICO_DEFAULT_LED_PIN, 1);

Modified the code for pico-w in the service routine
cyw43_arch_gpio_put(CYW43_WL_GPIO_LED_PIN, 1);

The board hangs after dropping the uf2 file in to board.  
USB also is not recognized on windows machine.

I found now that remoing the LED pin toggling part in service routine restores the repeating timer operation

globalkrishna, 2023-02-12:
part of the code is reproduced below:

bool timer_callback(repeating_timer_t *rt)
{
    refresh();
    time_up = true;
    if (++fsec > 99)
    {
        fsec = 0;
        sec_up = true;
        if (++tsec > 59)
        {
            tsec = 0;
            minute_up = true;
        }
    }
    //cyw43_arch_gpio_put(LED_PIN, fsec < 10 || (fsec > 20 && fsec < 30) ? 1 : 0);
}

in the main modified as
if (time_up)
          {
              time_up = false;
              cyw43_arch_gpio_put(LED_PIN, fsec < 10 || (fsec > 20 && fsec < 30) ? 1 : 0);
          }

modifed code works.

Is it wrong to call 'cyw43_arch_gpio_put' in interrupt service?

globalkrishna, 2023-02-12:
from the documentation I realize that

"Both the low level cyw43_driver and the lwIP stack require periodic servicing, and have limitations on whether they can be called from multiple cores/threads."

closed 2023-02-12 by globalkrishna