raspberrypi/pico-sdk #1244: lwip submodule should point to a tag or branch tip in the remote repository

ali1234, 2023-02-11:


ali1234, 2023-02-13:
I am not sure why it works now. It could be because the remote now points to the github mirror instead of savannah, and github might configure their servers differently. They never used to allow this either though.

kilograham, 2023-05-30:
um; i'm a bit confused, as this issue seems to refer repeatedly to `pico-extras` not `pico-sdk`... in any case it doesn't seem like #1369 is a suitable fix anyway, so won't fixing this for now - please re-open if there is still an issue

closed 2023-05-30 by kilograham

ali1234, 2023-05-30:
This issue was opened when lwip was a submodule of pico-extras. It is now a submodule of pico-sdk, and you moved the issue here when lwip was moved.

kilograham, 2023-05-30:
ha

kilograham, 2023-05-30:
anyways, i'm not sure there is much to be done, unless lwip makes a newer tag, right?

ali1234, 2023-05-30:
It does seem that this is no longer a problem. I am not sure why because github definitely is not advertising https://github.com/lwip-tcpip/lwip/tree/239918ccc173cb2c2a62f41a40fd893f57faf1d6 (you can confirm this with `git ls-remote`.) It seems to be a change in the behavior of the git command itself, to download the full remote even when you ask for a shallow clone - and then discard anything it doesn't need.