raspberrypi/pico-sdk #1247: Online doxygen reference still at 1.4.0, not 1.5.0

earlephilhower, 2023-02-12:
The online doxygen API docs at https://raspberrypi.github.io/pico-sdk-doxygen/ have not been updated to 1.5.0.

aallan, 2023-02-12:
These are deprecated as we're in the process of integrating them directly into the mainline documentation. See raspberrypi/documentation#2743 for details. I'd expect this to be live next week, potentially the week after, as a lot of us are on holiday for the February half-term school break right now. After that the [existing location](https://raspberrypi.github.io/pico-sdk-doxygen/) will redirect folks to the new pages.

closed 2023-02-12 by aallan

earlephilhower, 2023-02-12:
Thanks for the info.  Have a good break, you guys have earned it!

aallan, 2023-02-12:
I'd hoped to have it live before 1.5.0 was released, alas not! 🤷‍♂️

aallan, 2023-02-24:
Should go live on Monday, we just figured out the last of the bugs for the new site.