raspberrypi/pico-sdk #1248: v1.5.0 adds a dependency pico_cyw43_driver -> pico_stdlib

slimhazard, 2023-02-13:
While re-building code for SDK v1.5.0 (which compiles without error for 1.4.0), I encountered this error in the compile step:
```
/path/to/pico-sdk/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c:11:10: fatal error: pico/stdlib.h: No such file or directory
 #include "pico/stdlib.h"
          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/test-https-background.dir/build.make:1595: CMakeFiles/test-https-background.dir/path/to/pico-sdk/src/rp2_common/pico_cyw43_driver/cyw43_bus_pio_spi.c.obj] Error 1
```
Adding pico_stdlib to target_link_libraries() in CMakeLists.txt fixes it.

@kilograham said in a [forum thread](https://forums.raspberrypi.com/viewtopic.php?t=347338) that pico_cyw43_driver should not depend on pico_stdlib.

lurch, 2023-02-13:
Curiously, if I comment out that stdlib include, the SDK's kitchen_sink app still builds without error.

slimhazard, 2023-02-13:
> Curiously, if I comment out that stdlib include, the SDK's kitchen_sink app still builds without error.

Curiouser and curiouser, I tried that earlier with the app I'm trying to build, and there were link-time errors. Tried it again just now, and the app builds and tests just fine.

So either the universe moved in the meantime, or I had not properly cleaned up the build directory before re-compiling earlier. After a few more tests for sanity's sake, I can confirm that removing the `#include` solves the problem (and the universe is still working).

kilograham, 2023-02-13:
merged into develop

PR #1250 (2023-02-13): remove unnecessary headers - particular pico_stdlib.h which should not be included by libraries
fixes #1248