raspberrypi/pico-sdk #1255: sin(x) and cos(x) have increasing error for |x|>128

dpgeorge, 2023-02-14:
When using single precision float, `sin(x)` and `cos(x)` (and possibly `sincos` and `tan`) are accurate for `|x| < 128` (here, `|x|` means absolute value of `x`) but show increasing error as `|x|` grows beyond 128.

Here is a plot of the error of `sin(x)` as a function of `x` (compared to the single-precision float musl implementation of `sin`):

[image]

You can see that the error grows linearly beyond -128 / +128.

Looking at the code, there's a wrapper function that reduces the argument down to within +/- 128 and then calls the ROM code: https://github.com/raspberrypi/pico-sdk/blob/master/src/rp2_common/pico_float/float_aeabi.S#L637-L664 .  That reduction is probably introducing a small floating point error for each loop of the reduction (that's a guess, I did not investigate further).

(Note: this was found running MicroPython's `tests/float/cmath_fun.py` test, which fails on a Pico board due to the above problem.)

znmeb, 2023-02-14:
One should never call _sin (x)_ or _cos(x)_ with _|x|_ greater than 128! And code for _sin(x)_ and _cos(x)_ should take advantage of trigonometric identities to reduce the range of the argument. See https://realtimecollisiondetection.net/blog/?p=9, for example. The graph tells me the code may have range reduction but it's broken.

kilograham, 2023-02-15:
it is indeed broken ;-) oops.

dpgeorge, 2023-02-15:
@kilograham did you intend to close this?

kilograham, 2023-02-15:
nope

kilograham, 2026-11-10:
fix merged into develop

closed 2026-11-10 by kilograham