raspberrypi/pico-sdk #1256: log2, log10, acosh and tgamma don't return NaN when they should

dpgeorge, 2023-02-14:
The following functions evaluated at the given points return a finite or infinite result when then should return Nan:
- `log2(-INFINITY)` returns `-INFINITY` but should return NaN
- `log10(-INFINITY)` returns `-INFINITY` but should return NaN
- `acosh(-1.0)` returns `0.0` but should return NaN (it should return NaN for all arguments less than 1)
- `acosh(-INFINITY)` returns `INFINITY` but should return NaN
- `tgamma(-INFINITY)` returns `INFINITY` but should return NaN

I'm not sure exactly what the policy is for domain errors, but when `PICO_FLOAT_PROPAGATE_NANS` is enabled a lot of other cases of NaN are handled, so one would expect the above to also be handled correctly.

kilograham, 2023-02-15:
yes, there seems to be a mixture - initially the policy was intended to be PROPAGATE_NANs should take NAN input to NAN output, but we clearly have code in some places to fix this.

I'm surprised log(-INFINITY) isn't listed...

can you point me at the test for this in MP

dpgeorge, 2023-02-15:
`tests/float/math_domain_special.py` is the test.

That tests also see some strange behaviour of `math.asinh(0)` returning `-0.0` when it should be `0.0`, but I couldn't track that bug down because when I ran `asinhf(0.0)` in C code it returned `0.0` as expected.

dpgeorge, 2023-02-15:
> I'm surprised log(-INFINITY) isn't listed...

Surprisingly we don't have a test for this... I'm in the process of adding one but it seems that it does pass on rp2 because we handle `math.log(x[, base])` differently to log2 and log10: log explicitly checks that it's argument is positive.

dpgeorge, 2023-02-15:
To run the tests on MP:
```sh
$ cd tests
$ ./run-tests.py --target rp2 -d float
```
Then to see the diff of the expected (left) vs actual (right) results:
```sh
$ diff results/float_cmath_fun.py.exp results/float_cmath_fun.py.out 
113c113
< complex(-109.71, 55.602)
---
> complex(-109.72, 55.601)

$ diff results/float_math_domain.py.exp results/float_math_domain.py.out 
16c16
< sqrt(-inf) = ValueError
---
> sqrt(-inf) = -inf

$ diff results/float_math_domain_special.py.exp results/float_math_domain_special.py.out
7c7
< log2(-inf) = ValueError
---
> log2(-inf) = -inf
12c12
< log10(-inf) = ValueError
---
> log10(-inf) = -inf
26c26
< acosh(-1.0000) = ValueError
---
> acosh(-1.0000) = 0.0000
30c30
< acosh(-inf) = ValueError
---
> acosh(-inf) = inf
48c48
< gamma(-inf) = ValueError
---
> gamma(-inf) = inf

$ diff results/float_math_fun_special.py.exp results/float_math_fun_special.py.out
60c60
< asinh(0) = 0
---
> asinh(0) = -0
```
(The first one there is #1255.)

kilograham, 2026-11-10:
fix merged into develop

closed 2026-11-10 by kilograham