raspberrypi/pico-sdk #1260: How to actually set PICO_RP2040_USB_DEVICE_UFRAME_FIX=0?

czietz, 2023-02-15:
The [SDK 1.5.0 release notes](https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.0) mention the following new variable:
> Defaulted new PICO_RP2040_USB_DEVICE_UFRAME_FIX variable to 1 as a workaround for errata RP2040-E15. This sets the default value for TinyUSB's dcd_rp2040 driver's TUD_OPT_RP2040_USB_DEVICE_UFRAME_FIX flag. This fix is required for correctness, but comes at the cost of some performance, so applications that won't ever be plugged into a Pi 4 or Pi 400 can optionally disable this by setting the value to 0.

However: How do I actually set PICO_RP2040_USB_DEVICE_UFRAME_FIX to 0? I would have expected to be able to add it to my CMakeLists.txt like other configuration parameters:

```CMake
target_compile_definitions(my_project_name PRIVATE
PICO_RP2040_USB_DEVICE_UFRAME_FIX=0
)
```

However that results in a warning during the build process:
```
<command-line>: warning: "PICO_RP2040_USB_DEVICE_UFRAME_FIX" redefined
<command-line>: note: this is the location of the previous definition
```
... and checking the command-line shows that my project is still built with PICO_RP2040_USB_DEVICE_UFRAME_FIX=**1**.

Sadly, the SDK 1.5.0 documentation is (still?) missing, so I there isn't much documentation but the release notes.

PS: Also, the erratum RP2040-E15 is missing from the current RP2040 data sheet. Where I can find its description?
PPS: Good to see that my humble example apparently was used as part of the test case for the E15 erratum: https://github.com/P33M/rp2040-e15-testcase/commit/2707219c2a28fd7f601448bba68d53261b13a1b0

kilograham, 2023-02-15:
you have the right idea, but yes this is a bug.

for now use `TUD_OPT_RP2040_USB_DEVICE_UFRAME_FIX=0` instead (either via `target_compile_definitions` or in your tusb_config.h`

czietz, 2023-02-15:
Thank you. I can confirm that setting `TUD_OPT_RP2040_USB_DEVICE_UFRAME_FIX=0` via `target_compile_definitions` has the intended effect of disabling the workaround for erratum E15.

kilograham, 2023-02-15:
i have updated the release notes

kilograham, 2023-05-30:
changed to "documentation" as the workaround is actually the correct thing to do

lurch, 2023-05-31:
@kilograham AFAICT https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.0 and Appendix F (SDK 1.5.0. Release Notes) of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf and Appendix B (RP2040-E15) of https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf are all describing using `TUD_OPT_RP2040_USB_DEVICE_UFRAME_FIX` - what other documentation updates are needed to fix this? (or are the ducks all already in a row? :duck: )

P33M, 2023-06-14:
None?