raspberrypi/pico-sdk #1261: make_timeout_time_ms() is missing in v 1.5.0

NickDolfin, 2023-02-16:
I cannot compile  anymore without errors.
Apparantly make_timeout_time_ms() is missing
https://raspberrypi.github.io/pico-sdk-doxygen/group__timestamp.html

Or am I missing something?

czietz, 2023-02-16:
Yesterday, I compiled [my project](https://github.com/czietz/picowifi/blob/21fe5f1334d66ef0207b5752ee0448be06f552af/picowifi.c#L349) with SDK 1.5.0, and the project makes extensive use of make_timeout_time_ms(). In other words: That function is not generally broken/missing in SDK 1.5.0.

I suggest you provide a minimal reproducible example that shows the error that you're getting with SDK 1.5.0 but not with 1.4.0.

lurch, 2023-02-16:
https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.0 mentions that the headers were reworked in SDK 1.5.0, so maybe you were "implicitly" getting the header-file for `make_timeout_time_ms` included previously, and now you need to explicitly include it? :shrug:

NickDolfin, 2023-02-16:
Well, I think I made a mess of the toolchain.

I use platformio in Visual studio code.
But I have no idea wich version of the sdk I'm using. How do I find that out? (I've tried to check the files of the used sdk files, but there are no version numers in them).
Any idea on how to update the sdk?
The make_timeout_time_ms() does compile in this test file:

[image]

peterharperuk, 2023-02-16:
picotool will show you the version of the sdk a uf2 file was built with.

picotool info -a file.uf2

Build Information
 sdk version:       1.5.1-develop

https://github.com/raspberrypi/picotool

NickDolfin, 2023-02-16:
Thanks,

Error:

 
PS C:\Users\nick\Documents\PlatformIO\Projects\230216-195711-waveshare_rp2040_plus_4mb\.pio\build\waveshare_rp2040_plus_4mb> "C:\Users\nick\Desktop\Research\electronics\Raspberry pico\rp2040\picotool" info -a .\firmware.uf2
At line:1 char:77
+ ... top\Research\electronics\Raspberry pico\rp2040\picotool" info -a .\fi ...
+                                                              ~~~~
Unexpected token 'info' in expression or statement.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : UnexpectedToken

lurch, 2023-02-16:
Version numbers are also visible in https://github.com/raspberrypi/pico-sdk/blob/master/pico_sdk_version.cmake

NickDolfin, 2023-02-16:
Thanks,

I have version 1.4.0.
Can I just remove than directory and repalce it with 1.5.0? Or is there a special command sequence needed?

I use this in my platform.io: 

[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m

;[env:pico]
;board = pico

[env:waveshare_rp2040_plus_4mb]
board = waveshare_rp2040_plus_4mb

NickDolfin, 2023-02-16:
Perhaps earlephilhower is not yet ready to upgrade https://github.com/earlephilhower/arduino-pico/pull/1167 ?

Whats your idea?

lurch, 2023-02-16:
Neither PlatformIO nor Arduino are directly supported by Raspberry Pi, so you'll have to ask for help on their forums / GitHub repos.

NickDolfin, 2023-02-16:
That's a bummer.
I don't like the Arduino IDE for lack of lots of things. And the raspi standard toolchain does not support >2 MB platforms.'

lurch, 2023-02-16:
> And the raspi standard toolchain does not support >2 MB platforms.

It does, you just need to provide the correct `PICO_BOARD` value - see section 3.1 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf and section 2.6.1 (and/or Appendix D) of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

kilograham, 2023-02-28:
i think we've established that the original reason for this issue is bogus

closed 2023-02-28 by kilograham

cmarcante, 2023-05-12:
To solve this error; With version  1.5; into the file sd_spi.c add the line   #include "pico/time.h"

cmarcante, 2023-07-12:
Thanks; with "pico/time.h" I solved the problem.

zackfuchtel, 2023-09-22:
Thank you. Solved this problem for Semtech lorawan.c also.

ianarchbell, 2023-10-13:
Reason appears to be that spi.h in 1.5 does not include pico/time.h whereas it did in 1.4 so implicitly you were getting it included - now you need to explicitly include it.