raspberrypi/pico-sdk #1262: Instructions to Install openOCD are missing a required package

chriskot870, 2023-02-16:
I have a brand new Ubuntu 22.04 installation and was following the instructions on [Getting Started with Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf). In section 5.1 "Installing OpenOCD" I got a failure when doing the .bootstrap.
```
$ ./bootstrap
+ aclocal --warnings=all
+ libtoolize --automake --copy
+ autoconf --warnings=all
configure.ac:13: error: possibly undefined macro: AC_MSG_WARN
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:212: error: possibly undefined macro: AC_DEFINE
configure.ac:353: error: possibly undefined macro: AC_MSG_ERROR
configure.ac:652: error: possibly undefined macro: AC_MSG_NOTICE
```

I found this site regarding that autoconf error. [Autoconf undefined macro error message](https://stackoverflow.com/questions/8811381/possibly-undefined-macro-ac-msg-error). It claimed the following.

_I had this same issue and found that pkg-config package was missing._

So, I ran
`$ sudo apt install pkg-config`

After adding that package the bootstrap command ran fine. Also the configure command in the documentation and the make all succeeded without error.

I would recommend that the [Getting Started with Pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) document be changed to include pkg-config as a required package.

Change this line:
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev libusb-1.0-0-dev

To:
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev libusb-1.0-0-dev pkg-config

Thanks
Chris Kottaridis

aallan, 2023-02-16:
The instructions in the Getting Started book are intended for Raspberry Pi OS. Other versions of Linux may need more or different prerequisite packages.

closed 2023-02-16 by aallan

lurch, 2023-02-16:
The instructions in Appendix A, where we document installing openocd on a (non-Pi) Linux computer, **do** include a "Ubuntu and Debian users might additionally need to also install pkg-config." note.