raspberrypi/pico-sdk #1267: Setting WANT_HCI_DUMP to 1 in CMakeLists.txt target_compile_definitions() doesn't work

matsobdev, 2023-02-19:
Just like int the topic code below doesn't make BTstack to dump HCI comms:
```
target_compile_definitions(picow_bt_example_common INTERFACE
        PICO_STDIO_USB_CONNECT_WAIT_TIMEOUT_MS=3000
        #WANT_HCI_DUMP=1 # This enables btstack debug
        #ENABLE_SEGGER_RTT=1
        )
```
But, defining `#define WANT_HCI_DUMP 1` only in `btstack_config.txt` just works.

kilograham, 2023-05-26:
this seems odd; i assume you mean `bstack_config.h`
@peterharperuk is this meant to be a user setting; if so we should change it to be `PICO_BTSTACK_ENABLE_HCI_DUMP` or such

matsobdev, 2023-05-30:
To be honest, I don't know, maybe it was a mistake or something. It does work:
```
target_compile_definitions(hog_kbd PRIVATE
	TUD_OPT_RP2040_USB_DEVICE_UFRAME_FIX=0
	WANT_HCI_DUMP=1
	)
```
only in `CMakeLists.txt`. And yes, `bstack_config.h`.

peterharperuk, 2024-05-21:
Think this must be out of date or something. It works in the test code

closed 2024-05-21 by peterharperuk