raspberrypi/pico-sdk #1270: Help implementing CY43 gSPI

soypat, 2023-02-20:
Hello Rpi peeps,
I'm currently developing with TinyGo and wanted to implement CY43439 support to make use of the on board wifi capabilities. 

I've gotten [thus far](https://github.com/soypat/cyw43439), been able to read/write via an SPI bitbang implementation. The data I'm getting back from the chip is nonsense for the time being and it's very hard to debug without being able to hook up a logic analyzer to the CY's DIN/DOUT or GPIO24. So I was wondering if you guys had any SPI transaction samples used for the development of the pico-sdk's implementation. Anything helps really, images, tips and whatnot!

soypat, 2023-02-20:
Also wondering if there's firmware installed on the CY beforehand or if I'd need to flash it first...

lurch, 2023-02-21:
I guess you probably ought to be asking these questions at https://github.com/georgerobotics/cyw43-driver ?

soypat, 2023-02-22:
Update:
So I've managed getting the IO working on SPI! I have not been able to set the GPIO pins because the F1RxReady status check is failing. This is because I need to download the firmware to the CY43, no?

kilograham, 2024-05-19:
closing this, as this isn't an SDK issue

closed 2024-05-19 by kilograham