raspberrypi/pico-sdk #1277: Increasing the maximum timeout for Watchdog

DiwakerJha, 2023-02-23:
For my battery operated application, RPI PICO-W should wake up only every 20 minutes or longer. With maximum timeout for Watchdog timer, 8388 ms, I have to keep feeding much often than necessary? 

Is there a way to increase the maximum timeout of the Watchdog? or another way to mitigate possible hangs?

Thanks,

liamfraser, 2023-02-28:
This isn't really the intended use of the watchdog. I would recommend setting an alarm on the normal timer which is 32 bits. See https://github.com/raspberrypi/pico-examples/blob/master/timer/hello_timer/hello_timer.c for usage. You can then go into `__wfi();` and wait for the timer interrupt to go off.

bruelltuete, 2023-03-06:
the watchdog is clocked by clk_ref. You could change clk_ref to something slower, e.g. rosc (or maybe there's some clock division tricks you could do).

kilograham, 2024-05-19:
closing this as there isn't an intention to "fix" anything

closed 2024-05-19 by kilograham