raspberrypi/pico-sdk #1278: Can't use non-absolute offset for Bluetooth TLV storage area in PICO_FLASH_BANK

earlephilhower, 2023-02-23:
By default the 2 Bluetooth TLV stores are placed at the last 2 sectors of flash.

Unfortunately, in my own framework many people already have data in those last two sectors, so I need to move them elsewhere.  Dynamically would be best.  So I'm attempting to do something like creating a "const" variable that can be overwritten at the proper alignment in app code
````
const uint8_t __bluetooth_tlv[8192] __attribute__ ((aligned(4096))) = { 0 };
extern const uint8_t __flash_binary_start;

#define PICO_FLASH_BANK_TOTAL_SIZE sizeof(__bluetooth_tlv)
#define PICO_FLASH_BANK_STORAGE_OFFSET ((unsigned int)(__bluetooth_tlv - &__flash_binary_start))
````
(the undef/redef in the real app is handled in the `Cmakefile` and `btstack_config.h`, this is just a self-contained snippet for simplicity)

The problem is that the following static asserts fail due to the "non-constness" of the address-of operator.  The compiler can't know the offset of the variable, only the linker knows that, so the asserts fail.
https://github.com/raspberrypi/pico-sdk/blob/777cd52e21daf7c9a764e28611fceb3004ace488/src/rp2_common/pico_btstack/btstack_flash_bank.c#L13-L15

I would ask to either remove those `static_assert`s or bracket them with, say, a new marker `#if !defined(PICO_FLASH_BANK_DYNAMIC)`.  There doesn't seem to be any further dependency on the values being compile-time constants, only these assertions.

peterharperuk, 2023-02-24:
It sounds like we should just make it runtime rather than build time configurable?

kilograham, 2023-06-06:
merged into develop

closed 2023-06-06 by kilograham

earlephilhower, 2023-06-06:
Thanks, @kilograham and @peterharperuk !

PR #1293 (2023-06-06): Flash bank customisation
Defaults to true. Can be set to false if the flash bank offset is dynamically generated, e.g. to be inside the binary itself.

Fixes #1278