raspberrypi/pico-sdk #1280: assertion "irq_hander_chain_free_slot_head >= 0" failed

DwayneGit, 2023-02-24:
Hello, 

Im getting the error `assertion "irq_hander_chain_free_slot_head >= 0" failed: file "/home/robodev/pico/pico-sdk/src/rp2_common/hardware_irq/irq.c", line 215, function: irq_add_shared_handler`

This is the start of the main function and where the error is coming from:
```
    stdio_init_all();

    gpio_init(PICO_DEFAULT_LED_PIN);
    gpio_set_dir(PICO_DEFAULT_LED_PIN, GPIO_OUT);

    gpio_set_function(UART_TX_PIN, GPIO_FUNC_UART);
    gpio_set_function(UART_RX_PIN, GPIO_FUNC_UART);

    // // Turn off FIFO's - we want to do this character by character
    // uart_set_fifo_enabled(UART_ID, false);

    gpio_put(PICO_DEFAULT_LED_PIN, 1);

    gpio_set_irq_enabled(BRIDGE1_ENCA, GPIO_IRQ_EDGE_RISE, true);
    gpio_set_irq_enabled(BRIDGE1_ENCB, GPIO_IRQ_EDGE_RISE, true);
    gpio_set_irq_enabled(BRIDGE2_ENCA, GPIO_IRQ_EDGE_RISE, true); // Error occurs on this line
    gpio_set_irq_enabled(BRIDGE2_ENCB, GPIO_IRQ_EDGE_RISE, true);

    gpio_add_raw_irq_handler(BRIDGE1_ENCA, handler1);
    gpio_add_raw_irq_handler(BRIDGE1_ENCB, handler2);
    gpio_add_raw_irq_handler(BRIDGE2_ENCA, handler3);
    gpio_add_raw_irq_handler(BRIDGE2_ENCB, handler4);
    
    irq_set_enabled(IO_IRQ_BANK0, true);
```

Let me know if more information is needed and thanks in advanced.

kilograham, 2023-02-24:
```
// PICO_CONFIG: PICO_MAX_SHARED_IRQ_HANDLERS, Maximum number of shared IRQ handlers, default=4, advanced=true, group=hardware_irq
#ifndef PICO_MAX_SHARED_IRQ_HANDLERS
#define PICO_MAX_SHARED_IRQ_HANDLERS 4u
#endif
```

Add PICO_MAX_SHARED_IRQ_HANDLERS=5 or whatever to your target_compile_definitions in CMake

DwayneGit, 2023-02-25:
Thanks @kilograham that worked. Do you know why the default is 4 the trade offs for increasing this number?

kilograham, 2023-02-25:
it takes up space even if you don't use them, and 4 seemed like a reasonable number ;-)

closed 2023-02-25 by DwayneGit