raspberrypi/pico-sdk #1282: error: 'handle_sync_func_call' defined but not used

ygator, 2023-02-26:
The function handle_sync_func_call in the file pico-sdk/src/rp2_common/pico_async_context/async_context_threadsafe_background.c needs to be placed within #if ASYNC_CONTEXT_THREADSAFE_BACKGROUND_MULTI_CORE as the code that uses it is within the same #if 

```
#if ASYNC_CONTEXT_THREADSAFE_BACKGROUND_MULTI_CORE
static void handle_sync_func_call(async_context_t *context, async_when_pending_worker_t *worker) {
    sync_func_call_t *call = (sync_func_call_t *)worker;
    call->rc = call->func(call->param);
    sem_release(&call->sem);
    async_context_remove_when_pending_worker(context, worker);
}
#endif

```

kilograham, 2023-02-27:
indeed - fancy making a PR?

kilograham, 2023-05-30:
merged into develop

PR #1348 (2023-05-25): Add conditionalization for (potentially unused) struct and function
Add conditionalization for (potentially unused) struct and function (fixes #1282)
closed 2023-05-30 by kilograham