raspberrypi/pico-sdk #1292: undefined reference to `dns_setserver'

andmart, 2023-03-01:
I can't compile this and is a copy from pico-examples/pico_w/wifi/tcp_server/picow_tcp_server.c

```#include "lwipopts.h"
#include "pico/stdlib.h"
#include "stdio.h"
#include "pico/cyw43_arch.h"

int main() {
    stdio_init_all();

    if (cyw43_arch_init()) {
        printf("failed to initialise\n");
        return 1;
    }

    cyw43_arch_enable_sta_mode();

    printf("Connecting to Wi-Fi...\n");
    if (cyw43_arch_wifi_connect_timeout_ms("myssid", "12345678", CYW43_AUTH_WPA2_AES_PSK, 30000)) {
        printf("failed to connect.\n");
        return 1;
    } else {
        printf("Connected.\n");
    }
    cyw43_arch_deinit();
    return 0;
}
```

```
cmake_minimum_required(VERSION 3.13)

include($ENV{PICO_SDK_PATH}/external/pico_sdk_import.cmake)

project(teste C CXX ASM)

set(CMAKE_C_STANDARD 11)
set(CMAKE_CXX_STANDARD 17)

set(PICO_BOARD pico_w)

pico_sdk_init() 

add_executable(${PROJECT_NAME} teste.cpp)

include_directories(${CMAKE_CURRENT_LIST_DIR})

pico_add_extra_outputs(${PROJECT_NAME})

target_link_libraries(${PROJECT_NAME} pico_cyw43_arch_lwip_poll pico_stdlib)

```

```
build$ cmake .. && make
PICO_SDK_PATH is /home/andre/projetos/pico-sdk
PICO platform is rp2040.
Build type is Release
PICO target board is pico_w.
Using CMake board configuration from /home/andre/projetos/pico-sdk/src/boards/pico_w.cmake
Using board configuration from /home/andre/projetos/pico-sdk/src/boards/include/boards/pico_w.h
TinyUSB available at /home/andre/projetos/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/andre/projetos/pico-sdk/lib/btstack
cyw43-driver available at /home/andre/projetos/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/andre/projetos/pico-sdk/lib/lwip
Pico W Wi-Fi build support available.
mbedtls available at /home/andre/projetos/pico-sdk/lib/mbedtls
-- Configuring done
-- Generating done
-- Build files have been written to: /home/andre/projetos/pico-error/build
[  0%] Performing build step for 'ELF2UF2Build'
[100%] Built target elf2uf2
[  1%] No install step for 'ELF2UF2Build'
[  1%] Completed 'ELF2UF2Build'
[  4%] Built target ELF2UF2Build
[  5%] Performing build step for 'PioasmBuild'
[100%] Built target pioasm
[  5%] No install step for 'PioasmBuild'
[  6%] Completed 'PioasmBuild'
[  9%] Built target PioasmBuild
[  9%] Built target cyw43_driver_picow_cyw43_bus_pio_spi_pio_h
[ 10%] Built target bs2_default
[ 12%] Built target bs2_default_padded_checksummed_asm
[ 13%] Linking CXX executable teste.elf
/usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: CMakeFiles/teste.dir/home/andre/projetos/pico-sdk/lib/cyw43-driver/src/cyw43_lwip.c.obj: in function `cyw43_cb_tcpip_init':
cyw43_lwip.c:(.text.cyw43_cb_tcpip_init+0xb0): undefined reference to `dns_setserver'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/teste.dir/build.make:2262: teste.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1749: CMakeFiles/teste.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

andmart, 2023-03-01:
Pico-sdk: 

git log
commit f396d05f8252d4670d4ea05c8b7ac938ef0cd381 (HEAD -> master, origin/master, origin/HEAD)

cmake version 3.16.3

Ubuntu 20.04

gcc version 12.2.1 20221205 (Arm GNU Toolchain 12.2.Rel1 (Build arm-12.24))

peterharperuk, 2023-03-01:
I thought I'd fixed this already. There's an open PR here https://github.com/georgerobotics/cyw43-driver/pull/49

andmart, 2023-03-01:
#define LWIP_DNS 1 did the trick. Ty.

closed 2023-03-02 by andmart