raspberrypi/pico-sdk #1296: Where to find detailed doc for pico/cyw43_arch.h ?

shirshendubhowmick, 2023-03-02:
I am unable to find any documentation for `pico/cyw43_arch.h`

The official SDK [doc](https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf) doesn't include much details about it.

In the wireless support section of the doc it says:

> See the library documentation or the `pico/cyw43_arch.h` header for more details.

Please help me with where can I find this library documentation.

aallan, 2023-03-02:
> In the wireless support section of the doc it says:
> 
> > See the library documentation or the `pico/cyw43_arch.h` header for more details.
> 
> Please help me with where can I find this library documentation.

See https://www.raspberrypi.com/documentation/pico-sdk/networking.html.

shirshendubhowmick, 2023-03-02:
Thanks @aallan 

Just like to inform here, the I tried lot of googling for this doc. It looks the results available results in a 404.
For example:
https://raspberrypi.github.io/pico-sdk-doxygen/group__cyw43__driver.html

lurch, 2023-03-04:
Yes, we recently moved the location of the Doxygen API-documentation from `https://raspberrypi.github.io/pico-sdk-doxygen/` to `https://www.raspberrypi.com/documentation/pico-sdk/`, but hopefully in a few days google will update their links to point to the correct places.