raspberrypi/pico-sdk #1297: Bluetooth problems when overclocking

MichaelBell, 2023-03-02:
When overclocking to 250MHz, I can only connect to bluetooth audio the very first time after powering on the Pico W.  

After a reset (pulling RUN low), I can't reconnect.  At lower clock speeds this works fine.

Changing cyw43_bus_pio_spi.c as follows:
```
-    sm_config_set_clkdiv_int_frac(&config, CLOCK_DIV, CLOCK_DIV_MINOR);
+    const int clock_mhz = clock_get_hz(clk_sys) / MHZ;
+    const int clock_div = (clock_mhz < 170) ? 2 : (1 + clock_mhz / 85);
+    sm_config_set_clkdiv_int_frac(&config, clock_div, 0);
```
fixes it, but given this problem must be related just to some kind of reset, and I don't recall having issues with a 250MHz overclock doing WiFi with a Pico W, this fix seems extreme.

peterharperuk, 2023-03-03:
> can only connect to bluetooth audio

Which direction? Source or sink or both? i.e. Is the Pico playing audio or is it connecting to a speaker to play audio? The latter I assume?

MichaelBell, 2023-03-03:
Sink - Pico playing audio from my iPhone.  I'm running https://github.com/Gadgetoid/galactic-bluetooth-audio but with an overclock set in `main()`

peterharperuk, 2024-08-09:
I was going to mark this as fixed by https://github.com/raspberrypi/pico-sdk/pull/1521
But overclocking to 250 seems to work fine for me without having to adjust the pio divisor. It's somewhat curious that the a2dp_sink_demo now needs to be overclocked to work properly but that's another issue.
I don't now why pulling run low would affect things. I tried this and it reconnected fine.
I'll mark this as fixed as your original issue (having to hack the code to set the divisor) has now been fixed and I can't seem to repro the problem with a2dp_sink_demo. Please reopen if you disagree.

closed 2024-08-09 by peterharperuk

MichaelBell, 2024-08-09:
As this is in the realms of overclocking maybe it was related to the exact hardware I was running on.
Anyway, now there is the dynamic divider option for the CYW43 PIO that could easily be worked around if necessary so I agree with closing this issue.