raspberrypi/pico-sdk #1300: add file @specified address to uf2

underhood, 2023-03-04:
Hello, I somehow seem to fail at googling this.

What is the proper way to add arbitrary binary data to the uf2 image to be flashed.
What imagine is something like:
- add contents of a file "a.bmp" at address 1MB from flash start

so that this file is flashed together with firmware and then I can access it by something like
`const char *bitmap = (char*)XIP_BASE + 1024*1024`

lurch, 2023-03-05:
See #1071 and #1074 ?

kilograham, 2023-03-05:
1. simplest way to add image into your binary, is to use `xxd -i` to generate a C header
2. you can also mess with objcopy and friends to create a .`o(bj)` file you can include in your link
3. finally you can use `picotool` to load a separate binary
4. there is no UF2 tool i can see to combine two files (but you could make one in python relatively easily).

It sounds like in your use case you want 1 (since you just want it embedded anywhere) (or 2)

closed 2023-03-05 by kilograham

underhood, 2023-03-06:
@kilograham Thank you for reply the 1. would not work for my use case as I can not specify the exact location of the data on the flash. Reason I want to add file X at 1MB boundary in the flash is later I want the runtime to rewrite that location using the flash API. Therefore I wanted it to be at exact location and away from rest of the binary.

peterharperuk, 2023-03-06:
Why 1MB? Wouldn't 1 page be enough? This issue discusses an interesting technique for "reserving" an area in flash that you can update later https://github.com/raspberrypi/pico-sdk/issues/1278