raspberrypi/pico-sdk #1302: Compile errors on empty lwipopts.h

urbasus, 2023-03-10:
Compiling an example from pico-examples with empty lwipopts.h results in compile time errors:

```
In file included from .../pico-sdk/src/rp2_common/pico_cyw43_arch/include/pico/cyw43_arch.h:16,
                 from .../main.c:8:
.../pico-sdk/lib/cyw43-driver/src/cyw43.h:118:17: error: field 'dhcp_client' has incomplete type
  118 |     struct dhcp dhcp_client;
```

The surrounding code:

```
    #if CYW43_LWIP
    // lwIP data
    struct netif netif[2];
    struct dhcp dhcp_client;
    #endif
```

In lwIP dhcp.h we find this:

```
#if LWIP_DHCP /* don't build if not configured for use in lwipopts.h */
```

Expected behavior: lwipopts.h being a feasible default for any project using the SDK. Instead I naively copy the many opts used by the examples.

Suggested fix: From the looks of it some sections of the code should simply be suppressed similarly to how lwIP does it. Unclear whether padding is needed for binary compatibility should e.g. dhcp be disabled.

Observed on commit 2e6142b15b8a75c1227dd3edbe839193b2bf9041, tag 1.4.0.

peterharperuk, 2023-03-10:
Sounds like a duplicate of this https://github.com/georgerobotics/cyw43-driver/pull/49/commits/4be372ccf234e3ff4d790d92ec96b21ae1497ea8
It was merged a few days ago and will probably make it into the next sdk release.

peterharperuk, 2023-05-31:
I don't think somewhere along the way we've fixed the pico-sdk issues. And updating the cyw43-driver will fix almost all the remaining issues.

Depending on what you're doing, you'll probably have to define NO_SYS and when NO_SYS=0 lwip will complain unless you set the following...

#define NO_SYS 1
#define LWIP_SOCKET 0
#define LWIP_NETCONN 0

For free rtos you will get away with only defining the following. Maybe we should be defining this as we include sys/time.h in cc.h?

#define LWIP_TIMEVAL_PRIVATE 0

There's a final bug in the driver calling netif_set_hostname when LWIP_NETIF_HOSTNAME=0 which is the default.

peterharperuk, 2023-05-31:
Closing this as fixed (on develop) as I don't think it's feasible to do any more work (defining LWIP_TIMEVAL_PRIVATE would cause problems for existing code?)